espruino/BangleApps #15: Trouble setting "Year" in Settings, as the range is limited (-10  to +10)

VinceOPS, 2019-11-11:
Hi,

First, thanks for the Bangle.js!

Is there any reason why this `min` and `max` have been set? This leads me to not being able to set a Year higher than 1980 once my watch is restored to UNIX timestamp :sweat_smile: 

https://github.com/espruino/BangleApps/blob/6d2627a2f4463f4e23c60189b65ec9faa6e151a8/apps/settings.js#L252

I'd be glad to propose a PR if it is reasonable to change it.

shane-t, 2019-11-11:
As a workaround for now, I have found that if you set the year to max, then retreat from the menu, and then go back into it, you can keep incrementing it by 10 years (or use the GPS Time app which will set your clock without intervention)

gfwilliams, 2019-11-11:
Hah - yeah, I can't think of a good reason for that - thanks for the spot! A PR would be great!

VinceOPS, 2019-11-11:
@gfwilliams Hi there! Thanks for the prompt answer.
I'm a bit lost in the code right now, I'd like not to just propose a fix with some arbitrary values like `min: 1970` and `max: 3000`, but to add some tests, and understand where this `min`, `max`, `onchange` (etc) are used. Could you give me a few hints?

Thanks!

gfwilliams, 2019-11-12:
It's using getFullYear which will be returning values like 2019 - so you'd be totally safe with the min/max you suggest.

Probably the best bet is to load `BangleApps/apps/settings.js` into the IDE, then make your changes and click 'upload' - it should temporarily run your modified settings app so you can check if things work ok?

gfwilliams, 2019-11-12:
In terms of where they're actually used - this is the source code for the menu itself: https://github.com/espruino/EspruinoDocs/blob/master/modules/graphical_menu.js#L80

commit f04f6f62 (2019-11-20): just quickly fix #15
closed 2019-11-20 by gfwilliams