espruino/BangleApps #23: Special Chars for   "id": "aclock"

MaBecker, 2019-11-12:
just stumbled over special chars in the app name aclock.

>require('Storage').list()
=[
  ".bootcde",
  "+aclo\xEB\xEF%",       <---- guess some hidden chars in app.json
  "-aclock",
  "*aclock",
  "+sbat",  ...
  "+setting",
  "-setting",
  "=setting",
  "@setting",
  "*setting"
 ]

MaBecker, 2019-11-12:
Some more Infos:

* loaded Bootloader, than Settings and than AnalogClock 
* After resetting without loading any code and reloading the list looks correct.
* Display show "No Clock found" and menu entry has a "   Dead aclock" entry.

gfwilliams, 2019-11-12:
Have you tried reloading? I just looked in apps.json and there is nothing wrong there, so it could be there's some strange glitch. Could actually be because you've installed a new firmware over an old one. I'd be interested to see whether this ever happens again after a Storage.eraseAll

MaBecker, 2019-11-12:
Thanks, yes might be just did a extra eraseAll with  Banglejs.com - About - Remove all Apps.

>require('Storage').list()
=[  ]

so lets close it and reopen if it happens again

closed 2019-11-12 by MaBecker