espruino/BangleApps #35: Timezone from app Settings is not after BTN3

MaBecker, 2019-11-13:
Loaded app Settings and set "Time Zone " = 1
So now the app "Simple Clock" shows GMT+0001
Pressing BTN3 after loading another app, timezone is back to GMT+0000

Is there some startup code missing that loads settings?

gfwilliams, 2019-11-13:
What happens if you leave the clock for a minute? What I *think* happens is that the startup code does run (in `=settings/settings-init.js`, but it runs *after* the clock code does it's very first draw. Subsequent draws would be ok though.

I don't think there's going to be a big problem with swapping it around so that code (and the widgets in general) run before the clock

MaBecker, 2019-11-13:
Thanks, will change like this and than try again.

```
    // handle switch display on by pressing BTN1
    Bangle.on('lcdPower', function(on) {
        if (on) {
            drawWidgets();
            drawSimpleClock();
        }
    });
```

MaBecker, 2019-11-13:
Just checked, works great.
There is one side effect on widget-battery: Not always displayed in white.

Sample: Line 15  should be moved up after  line 9 to garantie white color
https://github.com/espruino/BangleApps/blob/541b40fc7f138b294b782e6f16199eed20e2df06/apps/widget-battery.js#L6-L16

closed 2019-11-14 by MaBecker

MaBecker, 2019-11-14:
Decided to add E.setTimeZone to the clock app I do.

```
 try{ E.setTimeZone(require("Storage").readJSON("@setting").timezone);} catch(e){}
````