espruino/BangleApps #45: Add linter and/or code formatter

ra100, 2019-11-15:
I think this could improve readability if all files were formatted the same, for example just by using `prettier`.

gfwilliams, 2019-11-18:
I'd be interested to hear other people's input on this, but my feeling is that these apps are all relatively small and they're self-contained.

I feel like the interests of being friendly to newcomers and getting a wide variety of apps probably outweighs having them all looking identical.

ra100, 2019-11-18:
After a bit of thinking you're right. If someone wants to use formatter, let them use their own style, but don't enforce it in any way.

johanbove, 2020-02-07:
We could provide an eslint plugin. In my experience eslint is one of the most used linting tools; See https://eslint.org/docs/developer-guide/working-with-plugins

gfwilliams, 2020-02-07:
Sounds like a good plan. jslint is what I currently use in the Web IDE's editor but eslint sounds preferable, and it might be possible to tweak it such that it complained about features that Espruino doesn't support (like destructuring), as well as usage of `save()` in global scope and stuff like that.

If it could be a separate repo I could hopefully look at bringing it into the Web IDE as well.

In terms of BangleApps a script that ran through and checked the JSON and JS files would be great, but I could look at that. Eventually it'll have to do some smarter stuff (checking the files referenced by apps.json, maybe even checking images are valid and the right size).

To be honest so far at least 50% of the contributions I've had haven't even tested the code on the emulator, let alone checked that BangleApps still works, so I think this isn't something that most people will actually run. However if we could set it as a Travis task at least there'd be an easy way to check if something was obviously broken before merging.

gfwilliams, 2020-02-28:
Just added 'sanitycheck.js' and enabled Travis support so it's a big improvement now. Already caught quite a few issues (icons, upload problems, etc).

It would be reasonably easy to add eslint support to the tool - it could probably get a list of globals from http://www.espruino.com/json/espruino.json (although it'd fail a lot of stuff right now because of the pre-release nature of the Bangle.js firmware)

rigrig, 2020-04-01:
> If someone wants to use formatter, let them use their own style, but don't enforce it in any way.

I use a formatter anyway, so would prefer if there was some recommended/suggested coding style.
So don't check it, but maybe put a hint somewhere in the documentation along the lines of
"Any coding style is fine, but if you are looking to pick one we recommend &lt;coding style&gt;"

gfwilliams, 2020-04-01:
Actually that would be a good one to add to espruino in general as we tend to use the same format for everything. For now:

* 1TBS style https://en.wikipedia.org/wiki/Indentation_style#Variant:_1TBS_(OTBS) but without the requirement for braces around single statements.
* Double spaces for indent (no tabs)
* lowercaseWithUpper variables/fields/etc (not a big deal)
* Capitalised objects
* UPPERCASE constants

The main reason for 1TBS is that Espruino's REPL bracket-counts whenever Enter is pressed to see if it should execute. See https://www.espruino.com/Bangle.js+Development#once-connected

gfwilliams, 2020-04-03:
Ok, there's now a page on what is suggested: http://www.espruino.com/Code+Style

johanbove, 2020-04-03:
@gfwilliams In the example code, did you leave out the braces in the "if" statement on purpose?
```javascript
if (1==1)
    print("Ok");
  else {
    print(1);
    print(2);
  }
```

gfwilliams, 2020-04-03:
It was vaguely intentional as I was trying to show you didn't have to use brackets, but it's a bit ugly and something I'd generally try not to do.

I'll change it to:

```
  if (1==1)
    print("Ok");

  if (1==1) {
    print("Ok");
  } else {
    print(1);
    print(2);
  }
```

detached, 2020-04-04:
We could add a .editorconfig file with some basic settings: https://editorconfig.org/

gfwilliams, 2020-04-06:
> We could add a .editorconfig file

Can do... To be honest this is the first I've come across those so I wonder how much they'd get used...

ikkala, 2020-05-26:
Made pull request, proposal for eslinting: https://github.com/espruino/BangleApps/pull/450

gfwilliams, 2020-05-26:
Thanks! Fixed with https://github.com/espruino/BangleApps/pull/450

closed 2020-05-26 by gfwilliams

ikkala, 2020-05-26:
The new pull request https://github.com/espruino/BangleApps/pull/451 forces spaces to 2 (and updates the whole code base to honor this rule).