espruino/BangleApps #46: [Suggestion] Code organization, folder per app

ra100, 2019-11-15:
Would it make sense to have a folder per app, for better navigation in the source?
example:
- `apps/animals/animals.js`
- `apps/animals/animals.json`

gfwilliams, 2019-11-16:
Thanks - yes, that could make sense. To be honest even with that it's going to get pretty hard to navigate, so I'm open to other ideas as well.

While having a single `apps.json` is a bit of a pain, there's something really handy about being able to test using GitHub pages on your own fork though.

ra100, 2019-11-16:
So for `apps.json`, would it make sense, to have that info actually in json of every app, and then just generating apps.json by simple script?
And going even further, what if every app had it's own repo and be linked to BangleApps just by apps.json? But that would mean, that apps could change without PR review to this repo, which sounds a bit unsecure.
Maybe just for now, some simple structure would be enough, and later, when it gets out of hands, to think more about it :)

gfwilliams, 2019-11-16:
> and then just generating apps.json by simple script?

Except that you could no longer just test it out using GitHub pages :(

While running a script seems fine to us, for a non-Node.js developer, installing Node.js then downloading and learning command-line git is quite a big learning curve.

Another option is to use something like https://github.com/gfwilliams/ThinGallery (so basically just scrape the website using client-side code), but that's slow.

> But that would mean, that apps could change without PR review to this repo, which sounds a bit unsecure.

Yeah, I mean if you don't want to do a PR you can always host your own repo. I think there's definitely some possibility of allowing proper third party stuff later, but especially for beginners I think having one repo with some level of vetting is a good idea.

ra100, 2019-11-16:
Yes, I didn't think about non-Node.js users. What about leaving it like it is, you'll just edit `apps.json` manually, for normal people :) and for lazy ones have js script, that would update it based on changes in apps. Buuut, right now, as I'm reading this, it looks like too much overhead. You'll have always the "check" in PR, where you'll see that apps.json was (or was not) updated.

Now I'm thinking how to combine good developer experience (with nice toolset, automated things and so on), but also keep it as simple as possible to add your own app.

ra100, 2019-11-16:
So how would moving to app folders look like.
- Have folder based on app id? Eg: `mclock` for Morphing clock, or `clock-morphing`?
- Should folder be specified in `apps.json`? `{"name":"+mclock","url":"clock-morphing/clock-morphing.json"},` or if it's the id, then create the path automatically, or add new property like `rootDir` or something with better name?

gfwilliams, 2019-11-17:
I'd say have the folder based on the 7 character name - which means we then easily avoid name clashes.

Probably not worth specifying the folder name in the JSON as it's just extra typing/opportunity for errors :)

gfwilliams, 2019-11-18:
Fixes with #50

closed 2019-11-18 by gfwilliams