espruino/BangleApps #51: [speedo] miles instead of km ?

chrisvel, 2019-11-18:
I opened speedo in my car and saw **66** while I was almost going with ~**110**km/h. 

I took a look and couldn't find something related in the code except that there is no "special" calculations in `speedo.js` but I can quickly multiply with `1.60934` (is 1km)  If that's the case.

gfwilliams, 2019-11-18:
Argh - thanks! I just looked into this and I was wrong about the data coming from the GPS receiver - it's in knots, not km/h. I've just fixed this on the firmware side so new firmware builds from http://www.espruino.com/binaries/travis/master/ should have it fixed!

closed 2019-11-18 by gfwilliams

chrisvel, 2019-11-18:
@gfwilliams After flashing firmware http://www.espruino.com/binaries/travis/master/espruino_2v04_banglejs.zip I can't get GPS to work, it just stays in `0 satellites` forever. 
Any ideas? Might be unrelated? 
I also tried pressing BTN1 + BTN2 to reset it but there was no change.

gfwilliams, 2019-11-18:
Can you just try leaving it outside for a bit with a GPS app running? Sometimes it can take a few minutes to get a lock if it hasn't been used for a while (or the battery ran down between times) - Smartphones can be a lot faster at getting a lock because they can ask the cell towers for time & position info to get them started.

chrisvel, 2019-11-18:
You are right but it took 5 minutes. At first dates it was taking no more than 1-2 minutes. I know cloudy weather can slow down the lock but can something else be happening or it's just my idea?

gfwilliams, 2019-11-19:
I bet if you try again now it'll be substantially faster?

Was the last time you tried it in Ireland? If so the GPS may have been working from the assumption you were somewhere else, which would have really slowed down the process of getting a lock.

chrisvel, 2019-11-19:
Indeed, in Ireland the lock succeeded sometimes in 1'. I have no idea why, but this is a question for some other place. Thanks.