espruino/BangleApps #58: Set Bangle.js Time issue

sebi5361, 2019-12-04:
Observations:
- When setting Bangle.js time using the Bangle.js loader the time displays correctly at first.
- However, if pressing for a long time on BTN3 the time gets displayed with **a time zone of 0**.
- To correct this, the user has to enter the correct time zone in settings, as it is 0 by default, even after setting the time using the Bangle.js loader.
- Then, if accessing the clock by pressing on its app from the application menu, the correct time taking into account the set timezone gets displayed.
- However if accessing the time by pressing for a long time on BTN3, the time gets displayed with a time zone of 0 at fist, and one has to wait till it gets updated (the next second, or next minute depending on the clock) to see the correct time taking into account the set time zone.

gfwilliams, 2019-12-04:
Thanks - the App Loader site should update the settings for timezone by the look of it.

Also the initial time being wrong is due to the bootloader running the 'widget' code after the clock code, with the settings startup code being treated as a widget. This could probably just be reversed without issues though

sebi5361, 2019-12-05:
Fixed. See commit [Bootloader tweaks to load widgets before clock](https://github.com/espruino/BangleApps/commit/24269067b9ddbd121275d0cd4514ef8d8b9c12f5).

gfwilliams, 2019-12-06:
Thanks! Although I'm going to leave this open for now as the App loader should set the timezone in settings as well (or it'll be lost on reboot)

sebi5361, 2019-12-06:
> the App loader should set the timezone in settings as well

Indeed.

Additionally I have noticed that the issue with the proper time zone being taken into account is not completely solved.

This can be illustrated easily by choosing _Simple Clock_ as the default clock (as it displays the time zone) and setting a none null time zone under settings:
When pressing on BTN3 for a long time, the clock gets displayed with a **time zone of zero** until it gets refreshed after about 10 seconds...

MaBecker, 2019-12-06:
> When pressing on BTN3 for a long time, the clock gets displayed with a **time zone of zero** until it gets refreshed after about 10 seconds...

Update the bootloader, the new version can handle this ;-)

sebi5361, 2019-12-06:
> Update the bootloader, the new version can handle this ;-)

Are you sure? I have uninstalled all the apps and reloaded them including the bootloader and still I get this issue: pressing BTN3 for a long time displays the time with a time zone of zero till it gets updated after about 10 seconds.

gfwilliams, 2019-12-06:
The updated bootloader is on GitHub but may not be live on bangle.com/apps yet

sebi5361, 2019-12-06:
Oh OK. Will wait then. Thanks

sebi5361, 2019-12-10:
It works fine now!
Only timezone tweak left to do.

commit 32a8136d (2020-01-17): Now set timezone setting from BangleApps (fix #58)
closed 2020-01-17 by gfwilliams