espruino/BangleApps #59: Reset via BTN3 causes error messages

MaBecker, 2019-12-04:
Looks like bootloader loads the clock app before loading  widgets.

Output:
 
Uncaught ReferenceError: "widget" is not defined
 at line 1 col 30
eval(require("Storage").read(widget))
                             ^
in function called from line 127 col 103
...re("Storage").read(widget)));
`

Switching to menu and back does not cause errors and widgets are loaded properly.

Any hints how to fix it?

Is this the expected load order?
* setting
* widgets
* clock if available

MaBecker, 2019-12-04:
>require('Storage').list();
=[
  ".bootcde",
  "+setting",
  "-setting",
  "=setting",
  "*setting",
  "+sbt",
  "=sbt",
  "+sbat",
  "=sbat",
  "+miclock",
  "-miclock",
  "*miclock",
  "@setting"
 ]
>

gfwilliams, 2019-12-05:
> `"@ setting" <-- added space`

That's extremely strange. It doesn't even seem possible since filenames are only 8 characters!

It's expected that the bootloader loads the clock before widgets, yes (and the settings app is in fact a widget) - there's another open issue at the moment (and you also spotted) the way timezone doesn't update immediately.

However, I assume the error comes from this line:

```
require("Storage").list().filter(a=>a[0]=='=').forEach(widget=>eval(require("Storage").read(widget)));
```

Since it's the only thing in the app store containing `eval(require("Storage").read(widget))` - and I just don't see how that can create the error, because it's a function with `widget` as an argument!

MaBecker, 2019-12-05:
Sorry, I added that space so it is not decode. as GitHub user

MaBecker, 2019-12-05:
Ok, started from scratch and loaded bootloader, widgets and apps with firmware.js technic.

closed 2019-12-06 by MaBecker

gfwilliams, 2019-12-06:
and it works fine?

MaBecker, 2019-12-06:
Yes, every thing works fine with the new bootloader version.