espruino/BangleApps #63: How to ease development/preview in emulator

gfwilliams, 2019-12-06:
The bootloader currently defines 'drawWidgets' and this means that if you just do a direct upload to the watch (or run something in the emulator) it's not there and causes an error. There may be more things in the future where this is an issue.

Maybe Bangle.js should just define those variables internally at boot time? That does obviously make stuff a bit less flexible though

sebi5361, 2019-12-11:
> Maybe Bangle.js should just define those variables internally at boot time?

This might be a good option to avoid having to encapsulate the `drawWidgets` function into a try block `try { drawWidgets(); } catch(err) {}`

gfwilliams, 2020-02-04:
Done now

closed 2020-02-04 by gfwilliams