espruino/BangleApps #64: App removal failed, Unable to decode "", got SyntaxError: Unexpected end of JSON input

sebi5361, 2019-12-06:
Often when I click on `Remove All Apps` I get such an error message in _Bangle.js loader_:
`App removal failed, Unable to decode "", got SyntaxError: Unexpected end of JSON input`
However apps get deleted, and I can press on the button again in doubt.

gfwilliams, 2019-12-06:
Hmm - do you get any other console messages?

Might be that the 'erase' call takes a while to complete, and it just times out waiting for a response

sebi5361, 2019-12-06:
When this problem occurs, the Chrome console shows:
```
puck.js:362 <BLE> Sending "\u0010Bluetooth.print"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ln(JSON.stringif"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "y(require(\"Stora"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ge\").eraseAll()|"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "|true))\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Unable to decode "", got SyntaxError: Unexpected end of JSON input
```
whereas in case of success it shows:
```
puck.js:362 <BLE> Sending "\u0010Bluetooth.print"
puck.js:362 <BLE> Received "true\r\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ln(JSON.stringif"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "y(require(\"Stora"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ge\").eraseAll()|"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "|true))\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Busy - adding Puck.write to queue
puck.js:362 <BLE> Executing {"type":"write","data":"\u0003\u0010reset()\n"} from queue
puck.js:362 <BLE> Sending "\u0003\u0010reset()\n"
puck.js:362 <BLE> Received "true\r\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Received "\u001b[?7l\r\n ____        "
puck.js:362 <BLE> Received "         _ \r\n|  __|_"
puck.js:362 <BLE> Received "__ ___ ___ _ _|_|___"
puck.js:362 <BLE> Received " ___ \r\n|  __|_ -| . "
puck.js:362 <BLE> Received "|  _| | | |   | . |\r"
puck.js:362 <BLE> Received "\n|____|___|  _|_| |_"
puck.js:362 <BLE> Received "__|_|_|_|___|\r\n     "
puck.js:362 <BLE> Received "    |_| espruino.com"
puck.js:362 <BLE> Received "\r\n 2v04 (c) 2019 G.W"
puck.js:362 <BLE> Received "illiams\r\n\r\n>"
```
Sometimes the empty string in `Unable to decode ""` is not empty but contains some commands instead.

gfwilliams, 2020-02-12:
Is anyone else seeing this issue with up to date firmware? It might be the Puck Storage has improved enough that it doesn't take long enough to cause any issues now

sebi5361, 2020-02-12:
I am still having this issue with the [latest firmware version](https://i.ibb.co/DM4T3cv/IMG-3363.jpg).
I get errors with the [app loader](https://i.ibb.co/h8spB7T/Image-2.png) and the [web IDE](https://i.ibb.co/mJ0gHSr/Image.png).
However, weirdly the app loader doesn't complain about [setting the time](https://i.ibb.co/yYMdwsc/Image-3.png).
Currently I am stuck as I cannot connect to the watch properly. I will try downgrading to an older firmware version that used to work properly...

(When investigating this issue, please remember that I used to have a different compiler version than yours that was responsable for some bugs with the bootloader on my DIY Bangle.js)

sebi5361, 2020-02-12:
I have downgraded the firmware to 2v04:
[image]
But still I get the same error messages with the App Loader and the WebIDE.
I also tried connecting to the watch when booted not loading any code, but it didn't help.
Note the two last lines about the GPS Time that didn't use to be there.

Pressing on BTN3 I get those errors displayed on the watch screen:
[image]
which might be related to 2v04 not being compatible anymore with the newest programs (maybe?) as I am stuck with the watch face (3x2 Pixel Clock) and have no possibility to access the Launcher.

The apps installed on my watch are: Bootloader, Launcher, Settings, 3x2 Pixel Clock, Battery Widget and GPS Recorder.

This bug started to occur after recording a GPS track (for about 15 minutes, successfully reaching some satellites) with the GPS recorder.
Just before I was able to connect to the watch and download an empty GPS track with
```
       <coordinates>
        NaN,NaN,NaN
      </coordinates>
```
only.
Maybe the GPS Recorder is the reason for all this mess?

Currently I am totally stuck and I might have to open the watch to use the SWD if I don't find a way to reset the watch...

sebi5361, 2020-02-12:
I reinstalled firmware 2v04.350 to get the Launcher to work.
I opened Settings to discover that, for any reason, *Programmable* was set to *off*.
Switching it back *on* allowed the App Loader to connect to the watch successfully.

However I confirm still that clicking on *Remove all apps* results in
[image]
despite all apps being removed successfully.

gfwilliams, 2020-02-13:
Yes, it sounded like your main issues were having `Programmable` set to off.

So, on latest firmware, what are the steps to reproduce?

* Erase all apps (even if error occurs)
* Install default apps
* Erase all apps - and error occurs

Because that doesn't happen for me at all.

sebi5361, 2020-02-13:
Today's [_App Loader_](https://banglejs.com/apps/) refuses to connect when clicking on _Remove all Apps_ or _Connect_. There is no pop up windows where to choose _WebBluetooth_ from that appears.

I will try the steps to reproduce the issue as soon as I find a way the let _Remove all Apps_ and _Install default apps_ work again.

gfwilliams, 2020-02-13:
Try again now. I accidentally added a typo before I committed.

sebi5361, 2020-02-13:
The App Loader connects fine.
- Erase all apps (even if error occurs)
Done. Error occurred.
- Install default apps
Got an error:
[image]
... and apps didn't install.
So I installed a bunch of apps from the Library manually (Bootloader, Default Launcher, Demo Loop, Morphing Clock, Settings, About) instead.
- Erase all apps
Error occurred still.

gfwilliams, 2020-02-13:
Does the error occur if you `Erase all apps` right after having previously done `Erase all apps`?

sebi5361, 2020-02-13:
No it doesn’t.

commit 5c3d1a48 (2020-02-27): Fix issues caused by Storage.eraseAll taking too long (fix #64)
Add progress bar and description to erase, uploads, and getting file info
closed 2020-02-27 by gfwilliams