espruino/BangleApps #68: Allow non-integer timezones

ryzokuken, 2019-12-18:
Hello! 👋 

I just got my Bangle.js from NodeConf EU this morning and started playing around with it. Apparently, it's not possible to set non-integer time-zones on the thing (I'm currently in UTC+5:30 and it became quite evident that this hasn't been addressed yet).

I thought changing the step size from 1 to 0.5 here (https://github.com/espruino/BangleApps/blob/master/apps/setting/settings.js#L101) will work, but maybe there's more nuance to it than that.

gfwilliams, 2019-12-18:
Thanks! Yes, changing the step should fix it. I'm away at the moment but if you add a pr I'm happy to merge it

MaBecker, 2019-12-18:
@ryzokuken, check this line too

https://github.com/espruino/BangleApps/blob/d306209da77cfc7c706f26054d59baaa422aff77/apps/setting/settings.js#L103

ryzokuken, 2019-12-18:
@MaBecker Yeah, I guess I'll just remove it.

PR #69 (2020-01-06): feature: all thirty-minute timezones
/cc @gfwilliams.

I tested it locally, it works. It'll be nice to work upstream on an alternate representation for this menu though (one that can go 5:30 instead of 5.5) but this should be as functional as anything.

Fixes: #68
closed 2020-01-06 by gfwilliams