espruino/BangleApps #73: Unable to access settings/menu with bootloader 0.03

ra100, 2019-12-28:
I've updated to 0.03 bootloader, and I'm not able to access menu (BTN2). When I uploaded initial bootloader, it's working fine.
Maybe something in 0.02 caused this.

gfwilliams, 2020-01-06:
Does anyone else get this? It seems I'd be getting loads of issues if it were the case. Maybe you could try updating your Bangle.js firmware if it's not up to date?

Can you still connect to your watch with the IDE? If so, maybe you could try doing BTN2 while connected and see if it outputs any error messages?

AkosLukacs, 2020-01-06:
Oh, yes, what firmware version do you have? Robin has a similar issue over the forums, but he is stuck at an earlier firmware version...

ra100, 2020-01-06:
I think I'll need to update FW, as I have nodeconf version.

MaBecker, 2020-01-06:
> Does anyone else get this?

Well, sometimes when using my iPhone this and other stuff happens, I guess this is because of a bad internet connection or the fact that on iOS devices BLE has to be used.

Issue: incomplete transfer.

ra100, 2020-01-07:
I've updated to latest bangle FW (espruino_2v04.84_banglejs.zip), and now it's working fine with 0.03 bootloader.

ra100, 2020-01-11:
So I'm closing this. As fq update helped.

closed 2020-01-11 by ra100