espruino/BangleApps #92: Installed settings from app cause Syntax error

MaBecker, 2020-02-07:
Hi, 

does https://banglejs.com/apps/ has the latest version?

Uncaught SyntaxError: Expecting a valid value, got ID
 at line 1 col 110
...false,"clock":null,"12hour":galse,"distance":"kilometer"}
                              ^
>require('Storage').list()
=[
  ".boot0",
  ".bootcde",
  "+boot",
  "+launch",
  "-launch",
  "+miclock",
  "-miclock",
  "*miclock",
  "+setting",
  "-setting",
  "@setting",
  "*setting"
 ]
>require('Storage').read("@setting")
="{\"ble\":true,\"dev\"" ... "nce\":\"kilometer\"}"
>console.log(require('Storage').read("@setting"))
{"ble":true,"dev":true,"timeout":10,"vibrate":true,"beep":true,"timezone":0,"HID":false,"clock":null,"12hour":galse,"distance":"kilometer"}
=undefined

gfwilliams, 2020-02-07:
Can you try re-installing?

I don't understand what happened here because `galse` isn't mentioned anywhere in the repository and if you look at the default settings file there's obviously no mention of it: https://banglejs.com/apps/apps/setting/settings-default.json

It's *possible* it's an internal error somewhere which cases an out of bounds JsVar lock/reference, so if you can find a way to reliably reproduce I can track it down.

MaBecker, 2020-02-07:
got a new one

 2v04.341 (c) 2019 G.Williams
Uncaught SyntaxError: Got '>' expected ','
 at line 1 col 12
{"ble":true>"dev":true,"timeout":10,"vibrate":true,"beep":tr...

gfwilliams, 2020-02-07:
Are you using the absolute newest firmware? I really need steps to reproduce this.

MaBecker, 2020-02-07:
> Are you using the absolute newest firmware?

Yes, just build this morning

```
=undefined
>process.env
={
  VERSION: "2v04.341",
  GIT_COMMIT: "1c3547a09",
  BOARD: "BANGLEJS",
  FLASH: 524288, RAM: 65536,
  SERIAL: "c7fa9624-75c07c51",
  CONSOLE: "Bluetooth",
  MODULES: "Flash,Storage,hea" ... "ow,graphical_menu",
  EXPTR: 536883676 }
> 
```

I will eraseAll() and start over again

MaBecker, 2020-02-07:
So after installing the apps

```
>require('Storage').list()
=[
  ".boot0",
  ".bootcde",
  "+boot",
  "+launch",
  "-launch",
  "+setting",
  "-setting",
  "*setting",
  "+miclock",
  "-miclock",
  "*miclock",
  "+sbt",
  "=sbt",
  "@setting"
 ]
> 
```

I switch the timezone  from 0 to 1 and that's it and various trouble happen....

```
 2v04.341 (c) 2019 G.Williams
Uncaught SyntaxError: Got '-' expected ','
 at line 1 col 63
...,"vibrate":true,"beep":true-"timezone":1,"HID":false,"clock"...
```

gfwilliams, 2020-02-07:
Do you install apps from https://banglejs.com/apps using the "Install Default Apps" button?

I just tried exactly that, changed the timezone, and everything is fine.

Can you try re-uploading after having done a `make clean`? It's possible your build is corrupted

MaBecker, 2020-02-08:
> Can you try re-uploading after having done a make clean? It's possible your build is corrupted

Yes, uploaded several time and "Settings" it often failing in my environment ( Mac OS X 10.15.3 with Chrome 79.0.3945.130)

Next is to update Chome to  80.0.3987.87 (Offizieller Build) (64-Bit) and run further test.

What environment are you using for all the firmware flashing and app transfering?

gfwilliams, 2020-02-10:
I'm using Linux and Chrome v79 at the moment. Could you do `require('Storage').read("@setting")` immediately after the upload and see if it's broken there? It seems like it would be the settings app itself that's corrupting the JSON but it'd be good to be sure

MaBecker, 2020-02-10:
Tested multiple deletes and add of app settings on multiple Bangle.

alway got this result:

```
>require('Storage').read("@setting").split(',');
=[
  "{\"ble\":true",
  "\"dev\":true",
  "\"timeout\":10",
  "\"vibrate\":true",
  "\"beep\":true",
  "\"timezone\":0",
  "\"HID\":false",
  "\"clock\":null",
  "\"12hour\":false",
  "\"distance\":\"kilometer\"}"
 ]
```

gfwilliams, 2020-02-10:
And is that broken? I can't see anything obviously wrong there?

MaBecker, 2020-02-10:
> And is that broken? I can't see anything obviously wrong there?

There was no error in that json when reading settings directly after flashing.

MaBecker, 2020-02-10:
Updated to espruino_2v04.343_banglejs.zip and reloeaded app  Settings

Tested Settings - Rest 

Console Message:
```
Uncaught Error: Unhandled promise rejection: ReferenceError: "setLCDTimeout" is not defined
>require('Storage').read("@setting").split(',');
```

https://github.com/espruino/BangleApps/blob/e1ddfe8ac4104213f6240e97272e02547dce199e/apps/setting/settings.js#L26

gfwilliams, 2020-02-25:
Is this ok now?

MaBecker, 2020-02-25:
Yes, so let’s close this.

closed 2020-02-25 by MaBecker