espruino/BangleApps #93: distance should use SI based units

MaBecker, 2020-02-10:
Using kilometer cause unreadable menu entry because of too much chars.

kilometer: km
miles: mi

gfwilliams, 2020-02-10:
thanks - that's a pain, I thought I'd fixed that.

I reckon 'mile' will be ok though? Probably less confusing than 'mi'

MaBecker, 2020-02-10:
Just tested, should distance unit match with distance?

```
>console.log(require('Storage').read("@setting").split(','))
[
  "{\"ble\":true",
  "\"dev\":true",
  "\"timeout\":10",
  "\"vibrate\":true",
  "\"beep\":true",
  "\"timezone\":0",
  "\"HID\":false",
  "\"clock\":null",
  "\"12hour\":false",
  "\"distance\":\"kilometer\"",
  "\"distanceunit\":\"mile\"}"
 ]
=undefined
> 
```

gfwilliams, 2020-02-10:
Wow, I must have been having a bad day - it shouldn't be creating `distanceunit` at all, just `distance`.

closed 2020-02-10 by gfwilliams

MaBecker, 2020-02-10:
Just updated, can you please switch kilometer to km.

[image]

gfwilliams, 2020-02-10:
It's already done (https://github.com/espruino/BangleApps/commit/3eb5d152672ebbcdab77a30a785edfb6907587bd), it's just that banglejs.com/apps wasn't up to date with GitHub yet.

Just updated banglejs.com/apps so if you try now it's fine