espruino/BangleApps #98: Some issues with the GPS Recorder App

sebi5361, 2020-02-12:
### 1.
The GPS Recorder App (its corresponding widget) modifies the color of the Watch Face.
Should the Watch Face App should be corrected or should it be the GPS Recorder App?
[image]

### 2.
Despite the Watch Face showing the proper time, the date and time of the recorded track are wrong.
[image]

### 3.
The last character of the .gprsc0 files is not displayed correctly in the App Loader (Windows 10).
[image]

gfwilliams, 2020-02-13:
1. Watch Face App - this should be changed in the watch app. `g.reset()` would be good to put before the draw call in each one.

2. Just fixed this - my quick attempt at rounding cropped it down to a 32 bit int :)

3. Just filed an issue for this - it's not to do with GPS: https://github.com/espruino/BangleApps/issues/99

4. Fixed now. The file still works fine though?

sebi5361, 2020-02-14:
Still a small bug with the track date displayed by the watch:
[image]

gfwilliams, 2020-02-14:
And that's for a new track? Not one of the old ones recorded before the fix?

sebi5361, 2020-02-14:
Indeed it is.
The date displays fine here since the fix:
[image]
...but not there:
[image]

sebi5361, 2020-02-21:
I think this issue just has to do with `Date().getTime()` being in ms while `getTime()` is in seconds.
Thus `Date(getTime()*1000)` is the proper formula.

gfwilliams, 2020-02-24:
I think in this case it might just be the `0|` - same fix I made for the widget, but just in the app.js: https://github.com/espruino/BangleApps/commit/619786cae00d8d536324afacc9ac79c943ebc31d#diff-80e5d79fdecd6bca9027b76a523689acL46

closed 2020-02-24 by gfwilliams

sebi5361, 2020-02-25:
Hi Gordon, I have checked your fix and am still getting the wrong date and time on the watch~~

gfwilliams, 2020-02-25:
Ok, thanks. Fixed (and tested) for reals this time - I should have wrapped it in a `parseInt`

sebi5361, 2020-02-25:
Indeed it works fine. However there is a Time-Zone hour difference between the time read on the watch and the one when downloading the file. It might be due to using `toISOString()`.

gfwilliams, 2020-02-26:
Hmm, that's an interesting one. The time itself is just sent over as a raw value of milliseconds since 1970 (standard unix timestamp) - on the watch it's interpreted using the current timezone that's specified in settings, and I'd assumed the same would happen on the PC!

commit 619786ca (2020-02-13): Ensure GPS time is logged correctly and fix output indentation (fix #98)
commit f7024407 (2020-02-24): Fix GPS time display in gpsrec app (fix #98)
commit cea2c9f7 (2020-02-25): Properly Fix GPS time display in gpsrec app (fix #98)