espruino/BangleApps #100: Pedometer widget causing battery drain

sumitkm, 2020-02-13:
Looks like the pedometer widget is causing a bit of battery drain. It managed to drain my BangleJS completely in about 12 hrs. 
I noted a separate thing, the screen came on a LOT more easily when the pedometer widget was installed. I initially thought it was due to me resetting the watch and getting new versions of the apps. But after I uninstalled the pedometer widget the 'screen on' behaviour went back to it's old self.

I just noted this line of code and am wondering does this kick the screen back to on ?
https://github.com/espruino/BangleApps/blob/master/apps/wpedom/pedometer_widget.js#L44

In that case, maybe we should handle the `lcdPower` event and only call `draw` when the screen is turned on ?

gfwilliams, 2020-02-13:
Thanks for this - that's really helpful!

Yes, it turns out it was `draw()` which in turn called `g.flip()` - it was turning your screen on for every step, which would have killed your battery in no time!

All fixed now - I'll update bangle.com/apps shortly!

closed 2020-02-13 by gfwilliams

sumitkm, 2020-02-13:
Awesome! I also noted that the Pedometer lost the step count a couple of times when I went to the `Settings` app and back, is that because I am having an old firmware that doesn't support `E.kill` event yet ?

gfwilliams, 2020-02-13:
> is that because I am having an old firmware

Possibly. If it showed the right step count when you returned to the clock but *then* lost it the next time you made a step, that was a bug that I have now fixed too :)

bangle.com/apps just updated, so if you try now things should be better

commit 6e6d078a (2020-02-13): Pedometer now doesn't turn the screen on for each step (fix #100)