espruino/BangleApps #102: Bluetooth connection state messes up apps

gfwilliams, 2020-02-14:
The console was moved permanently onto Bluetooth, but I moved it back to 'auto' (so it goes to the screen if BT is disconnected) recently as it seemed good for debugging (so any errors get shown to the user).

But this means that BT connection state scrolls the screen and messes stuff up.

So we could:

* Add a 'debug' option that, when false (default) moves the console to Bluetooth. When true you get the scrolling, but as you're developing you put up with it :)
* Add an option to Espruino to disable the Connection State messages
* Change apps/widgets to clear their backgrounds all the time, so scrolling was less of a big deal
* Leave the console on Bluetooth permanently - but then you don't see app erros

conoro, 2020-02-24:
That connection state message is so annoying when you are not in dev mode. Ditto any console type errors/messages. We've held off from sending watches to a few of our new employees as the immediate GPS error message on the watchface just makes it look like it's broken.

So I'd be a fan of option 1. When I'm not in dev mode, I never want to see any messages on the watch screen, including JS errors etc.

gfwilliams, 2020-02-24:
Thanks - option 1 is an easy fix at least, but it does make it more likely that apps will get included that have errors in them :)

conoro, 2020-02-24:
Yeah true! We just need a CI/CD pipeline on the emulator and machine vision to interpret the error messages ;-)

gfwilliams, 2020-02-25:
Fixed with https://github.com/espruino/BangleApps/commit/9a76d6d8003e1ab98835ff6eccc7c7a5df51710e

closed 2020-02-25 by gfwilliams