espruino/BangleApps #104: GPSTime broken

conoro, 2020-02-17:
Firmware download from site today along with latest code from BangleApps.

Similar issue with apps/firmware last Wednesday too. But I was also getting GPS time errors on main settings screen last week. And only satellite image showing on screen in app last week, not satellite count.

[image]

conoro, 2020-02-19:
So everything works fine just after I do "install default apps", as the watch has received an accurate time from the PC. But if I reboot, it obviously reverts to Jan 1 1970, and then all of the errors described above appear. As it's a Firmware issue I should probably close this and open it on Espruino repo?

gfwilliams, 2020-02-24:
No, that's fine - I think it's an app issue. Newer firmwares set `fix.time===undefined` if the GPS doesn't have a time yet, and I imagine that's causing the issue.

gfwilliams, 2020-02-24:
Also, new firmware automatically grabs GPS time on reboot (even without the app) if the GPS time is set

conoro, 2020-02-24:
Thanks Gordon. Something is also logging "GPS receiver's time not set" to the screen too after a reboot. Trashes the watchface.

gfwilliams, 2020-02-24:
Thanks - yes, that'd be the code above. I guess that logging could just be removed. Also related to https://github.com/espruino/BangleApps/issues/102 so I'm interested to see what you think the best fix might be

conoro, 2020-02-24:
Option 1 in #102 for sure. Should definitely be different behaviour between "I'm using this as a watch day to day" vs  "I'm developing on the watch". For the former, it should be as close to a consumer experience as possible.

commit f3efada3 (2020-02-24): 0.03: Fix time output on new firmwares when no GPS time set (fix #104)
closed 2020-02-24 by gfwilliams