espruino/BangleApps #110: Both GPSTIme and GPSInfo broken after watch restart until time is set from App Loader

conoro, 2020-03-04:
GPSTime shows only the satellite animation after watch restart. 

GPSInfo only shows part of the screen from the Welcome App after watch restart. 

Both work ok if I set the time from the App Loader first.

[image]
[image]
[image]
[image]

gfwilliams, 2020-03-04:
Thanks for this - just figured out what happened. I had this on Monday and figured it was an issue with uploading corrupt firmware (now fixed)

It's the code that attempts to update the time from GPS - it was starting at the same time as the GPS app, then shutting down the GPS. Only happens at first boot now so it should be sorted.

closed 2020-03-04 by gfwilliams

conoro, 2020-03-04:
Thanks Gordon that sorted it out. One thing I've just noticed, not sure if it's new or I missed it before, but it now displays "Searching for GPS Time " on the screen any time I switch between apps (i.e. long-press BTN3). Is it actually searching and, if so, is there any battery implication for this if it's indoors and never gets the time?

gfwilliams, 2020-03-04:
Yeah - before it was doing it, but it never told you.

There would be a (small) battery effect (40mA for 1 second). However the second you set the time correctly *or* left the GPS running enough to get a lock, you wouldn't see it at all.

conoro, 2020-03-04:
Ah that's perfect. It's time for me to flash a bunch of older watches with the latest!

gfwilliams, 2020-03-04:
We're just in the process of adding translation, so in a few days you might get that built in as well if you want to wait.

commit b93c0943 (2020-03-04): Fix GPS apps if time not set in GPS (fix #110)