espruino/BangleApps #121: Any idea why charging banks lower screen part ?

MaBecker, 2020-03-09:
start charing, erases the lower screen section

[image]

next clock update fixit it
 
[image]

same happens when unplug from charing

gfwilliams, 2020-03-09:
It's because of the `Bangle.drawWidgets()` which now erases the 'widget areas' at the top and bottom of the screen. I've made some tweaks so it avoids that, but even so if someone has a widget at the bottom the screen it'll erase that corner for you.

`Bangle.drawWidgets()` wouldn't normally be called, but it *is* because now the length of the battery indicator changes depending of whether charging or not

closed 2020-03-09 by gfwilliams

MaBecker, 2020-03-09:
Thanks,

> 'widget areas' at the top and **bottom** of the screen

bottom area : last 24 pixel, 

I guess that's a new nice feature that will allow more  widget - cool - thanks!

As a consequence: new app section is (0,24,240,215) - correct ?

gfwilliams, 2020-03-09:
Yes, that's right - there has always been a bottom widget area, just nobody used it yet :)

MaBecker, 2020-03-09:
Is it possible to specify section I like to use and if how?

MaBecker, 2020-03-09:
https://github.com/espruino/BangleApps#graphic-areas  need's some polish ;-)

gfwilliams, 2020-03-09:
Ahh, yes - there is info on doing widgets on https://github.com/espruino/BangleApps#widget-example now