espruino/BangleApps #122: Run in simulator button missing when accessing https://banglejs.com/apps

sebi5361, 2020-03-09:
For an unknown reason, there are no buttons to run apps in the simulator when accessing https://banglejs.com/apps/ using Chrome on my Windows 10 PC.
However if I fork BangleApps repository, clone it and run it on my local server, the buttons to run the apps in the simulator show up.

gfwilliams, 2020-03-09:
Can you try pulling up the Chrome dev console then click-clicking on 'reload' and choosing 'Empty Cache and hard reload'?

sebi5361, 2020-03-09:
I have done what you told me and found the answer: it is related to my Adblock Plus extension:
If I disable it on the banglejs.com/apps/ website, the buttons appear as expected.
I will add banglejs.com to the Adblock Plus extension white list not to get this issue.
I believe this issue might concern many other people as many of us use ad blockers...

However I am still having a bug:
If I click on such a button I get this error message:
// Error loading https://banglejs.com/apps/#apps/mclock/clock-morphing.js
after following this link:
https://www.espruino.com/ide/emulator.html?codeurl=https://banglejs.com/apps/#apps/mclock/clock-morphing.js&upload
The #apps in the link looks suspicious.

sebi5361, 2020-03-09:
I have noticed that I don't get this issue when clicking on the button when I am on https://banglejs.com/apps/
However if I am on https://banglejs.com/apps/# I get the issue.

gfwilliams, 2020-03-09:
Wow, good spot about the '#' thing. I'll get that fixed.

wrt. AdBlock - if you figure out what's wrong it'd be great if you could issue a PR. *Or* if there were a way to detect AdBlock and put a banner at the top of the screen telling people about it, that'd be cool.

sebi5361, 2020-03-09:
I am not a "real" programmer and I believe this is out of my reach... Sorry
(I barely know how to play around with JavaScript by adjusting your code to my needs~~)

On my local server when clicking on the simulator button, this link opens:
https://www.espruino.com/ide/emulator.html?codeurl=https://sebastienkrut.synology.me:447/apps/demoapp/app.js&upload
and I get this error message:
// Error loading https://sebastienkrut.synology.me:447/apps/demoapp/app.js
However the link to the file seems correct and I can access it.
Do you know the reason for this problem? Does it have to do with the :447 port in the link?

gfwilliams, 2020-03-09:
Ok, sorted (at least the link issue)

It probably doesn't work for you because you need 'Cross Origin' headers on your server - check out https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS

closed 2020-03-09 by gfwilliams