espruino/BangleApps #128: Look's like Battery Level Widget is not updated when removing charge cable

MaBecker, 2020-03-16:
Version 2v05, removed all apps and tried with the latest version.
Bootloader was 9 and is now 10

Cable is removed but 

[image]

or 

[image]

or

[image]

Let me know if you need more details or tests.

gfwilliams, 2020-03-17:
Ahh - you mean because the battery widget resizes and things move around, and we now don't clear the whole top and bottom row because it interferes with the analog watch face, part of the widgets in the previous position gets left behind?

I think this'll require a change in Espruino firmware itself :( I guess the simple solution is just to move to erasing the entire top/bottom row if there were any widgets there. It'd make the code simpler anyway.

MaBecker, 2020-03-17:
This is my workaround: just do a reset with BTN3 and everything is fine

gfwilliams, 2020-03-17:
In some watch faces (the ones which do `g.clear()` when the LCD is turned on) it'll clear up pretty quick.

We could just modify the bootloader to overwrite `Bangle.drawWidgets` with the older version