espruino/BangleApps #134: Settings should allow changing sensitivity of LCD auto-on

gfwilliams, 2020-03-20:
This is trivial to do, and I think would be helpful for a lot of people

gfwilliams, 2020-03-30:
also wakeOnBTNx: http://forum.espruino.com/conversations/345297

detached, 2020-04-03:
I can imagine two different ways for implementing the sensitivity settings:
1. exposing the relevant [options](https://banglejs.com/reference#l_Bangle_setOptions) as they are in the menu
2. hiding the options behind a "Auto-On Sensitivity: low, middle, high"

Do you have a preference?

gfwilliams, 2020-04-03:
From a user point of view I feel like low,middle,high is sensible, but then I also see that being a minefield of bug reports saying 'low is too high', etc. I think in a lot of cases it might be something that needs just a 10-20% tweak to work better.

I think maybe as `setOptions` says `accelerations are used they are in internal units, where 8192 = 1g`, we could make `twistThreshold` measured in g, and adjustable by 0.01g? It then makes it pretty easy to tweak from the default of 0.1, and hopefully not too scary :)

Also if a lot of people find that one setting is better, it would be easy to translate that into values that could be fed back into Espruino's defaults :)

PR #225 (2020-04-06): Add lcd wake-up settings
Resolves #134 by adding a LCD Wake-Up menu.

The wake-up settings are written to the settings json and reloaded after boot.
closed 2020-04-06 by gfwilliams