espruino/BangleApps #141: latest upgrade clock not visible

giacomo, 2020-03-25:
Version 0.03

gfwilliams, 2020-03-25:
Can you provide more info on how you got the problem? Did you try installing again?

I just did 'Install default apps' here to install everything and mclock 0.03 works great.

giacomo, 2020-03-25:
installed reinstalled tried everything.
now I've installed again and the screen remains black only the widget starts. today received and updated all installed apps. 

and version 0.02 has run out of the box but todays update didn't work.

installed locale de
update morphing clock to 0.03

gfwilliams, 2020-03-25:
Ok, thanks - the `de` locale was the trick.

Please can you try re-installing the `de` language now? That should have fixed it.

closed 2020-03-25 by gfwilliams

giacomo, 2020-03-25:
@gfwilliams good work! awesome, thanks :rocket: