espruino/BangleApps #142: drawString with Degree(°) Sign will not work

TheAkki, 2020-03-25:
When i call  `g.drawString("015°00.0000' O", 0, 0, true);` the degree sign is missing. There is no empty space.
I use Font "Vector12".

MaBecker, 2020-03-25:
better post in forum  [JavaScript](http://forum.espruino.com/microcosms/116/)

gfwilliams, 2020-03-25:
I believe this is expected. Does font `6x8` work? Because of the limited memory, only 6x8 font contains the full 256 chars. Vector and 4x6 contain only ASCII digits.

TheAkki, 2020-03-25:
Yes, 6x8 it working. 
For the first try the size was crazy. But now i think, that the size is for vector fonts: pixels and for bitmap fonts: the multiplier of height.
So it works for me.

gfwilliams, 2020-04-09:
Closing this as it's expected. I have attempted to document what chars there are at http://www.espruino.com/Fonts

There was some talk a new vector font at https://github.com/espruino/Espruino/issues/1721 and hopefully we could get a degree & euro sign in that

closed 2020-04-09 by gfwilliams