espruino/BangleApps #144: Enhancement: gadgetbridge calendar sync

detached, 2020-03-26:
I would like to propose an enhancement of the gadgetbridge widget and use this issue for coordinating development efforts.
Gadgetbridge supports syncing the phone calendar to a device:
https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/externalevents/CalendarReceiver.java

How about storing the calendar events as json on the banglejs (https://banglejs.com/reference#l_Storage_writeJSON) and subsequent develop a calendar app that can read this events?

gfwilliams, 2020-03-26:
Yes, we could definitely do that. It could be added as another form of data from Gadgetbridge.

However, is the idea that your watch could then alarm you even when it wasn't connected to Gadgetbridge? Because if it's connected then you should get calendar notifications pushed from the phone anyway?

detached, 2020-03-26:
I had an overview of all calendar events of a day in mind, not a reminder.
Like this https://support.apple.com/guide/watch/calendar-apd1b51754cc/watchos, but maybe first in the phone->clock direction.

It would take three steps:
* store gadgetbridges calendar events 
* develop calendar app
* activate calendar syncing in gadgetbridge for banglejs

gfwilliams, 2021-12-06:
Wow, yes - that'd be a great use for this!

glemco, 2022-05-23:
Hi all, I just got my bangle and I'm quite satisfied, I chose it mainly as it looked the only feasible alternative to have calendar sync on something FOSS (non WearOS). Did someone work on this issue so far?
I had a check at the code for both GB and bangle and I have some ideas, if there's nothing ready out there I can start have a deeper look on it, if time allows.
Is there something special I should care about besides the useful 3 steps @detached suggested?

gfwilliams, 2022-05-23:
It'd be great if you wanted to get stuck in. Yes, I'd say those 3 steps. Maybe first look at getting calendar output in the Gadgetbridge code, so it'll push data out for the next day or so along the lines of http://www.espruino.com/Gadgetbridge

Did you intend that it'd remind you, or just that you'd be able to see the events for the day? If it's actually going to remind you then I guess you might need to use the `sched` library to handle all the scheduling for you

glemco, 2022-05-23:
Thanks for the response!
I checked that and how it is implemented in GadgetBridge, I was thinking to somewhat get inspiration from the Pebble timeline and just use the encoding for bangle (the one explained in your link).
What I'd like to see is an agenda-like app, maybe limiting the number of events not to use too much memory (I'm not sure what is GB doing for this), sure alarms can be handy but I guess if the phone is connected they are already received as notifications.

gfwilliams, 2022-05-23:
Ok, that sounds great - and I think it'd be nice and easy to implement in a way that was all backwards compatible

glemco, 2022-06-20:
Quick update on the status of this, on the bangle side everything is in place.
I submitted a pull request on GadgetBridge's officail repo and I'm still waiting. This will probably take a while as GB is undergoing a refactoring in the way it handles devices and the calendar functionality looks broken.
Basically GB should now be able to handle more devices at the same time, but the CalendarReceiver component requires a device object and now is initialized with null.
Until they figure how to handle that, I think I can only wait.

gfwilliams, 2022-06-27:
Ok, thanks for the update!

closed 2023-08-23 by detached