espruino/BangleApps #147: Deleting all clock apps prevents the start of the app menu

profriemke, 2020-03-26:
If all existing clock apps are deleted (e.g. accidentially) the watch only shows a message that no clock app is present. The app menu is then not accessible.
Suggestion: If no clock is present, the app menu should be shown as standard.

gfwilliams, 2020-03-26:
Thanks - that sounds like a great idea.

brainfart-be, 2020-03-29:
Seeing the launcher directly felt weird, i decided to just add the BTN2 click to open the menu.

MaBecker, 2020-03-29:
I suggest to start the About app.

brainfart-be, 2020-03-29:
I personally removed it from my watch, i would end up with the same issue.
We can maybe make the smallest watch app possible with a message saying "to install a clock go to https://banglejs.com/apps#clock" or something like this.
(The #clock part is not yet live but the features was merged already)

gfwilliams, 2020-03-30:
I think your current solution is nice and tidy for the moment. If someone doesn't have a clock it's extremely unlikely they'll have the About app I guess :)

closed 2020-03-30 by gfwilliams