espruino/BangleApps #151: GadgetBridge: Notification never stops?

fungus75, 2020-03-26:
I've just configured my Android with GadgetBridge and the Bangle-Smartwatch.
As soon as I get an notification on my android - i.E. WhatsApp - I get the same on my Bangle. Great. But: the Bangle vibrates and never stops. I do have to push a button; best case: Button 3 for round about 3 secs.

I guess this could not be the requested feature?
I'm using the current GadgetBridge on Android from F-Droid and the newest available GadgetBridge-App for Bangle.

gfwilliams, 2020-03-26:
Did you try tapping the screen to dismiss the notification?

fungus75, 2020-03-27:
Yes. Then the "Messagebox" disappears, but the watch keeps vibrating.
I'm specially thinking of this: what if during the night, when I'm sleeping I'm getting a notification? Does the watch will vibrate that strong that I will wake up?

From my point of view, the whats never stops vibrating until I press a button (Button2 and Button3 works).

Currently I have waited up to 30 Seconds.

fungus75, 2020-04-01:
Looks like this issue only occurs when using the centerclock-Watchface.
With Mixed Clock, for example, it works perfect: one short vibrate, not more.

gfwilliams, 2020-04-02:
Thanks - that's really helpful for narrowing it down!

gfwilliams, 2020-04-02:
Does this happen all the time? Just tried it here, and it only happens on the first notification after GadgetBridge connects. Subsequent notifications are fine.

I've narrowed it down to Gadgetbridge sending a `setTime` command when it connects. When the new Gadgetbridge release comes out you'll just be able to turn off 'Time Sync' and it'll work fine.

I'm still unsure why Centerclock in particular has this problem though!

gfwilliams, 2020-05-15:
Does this still happen? It might be fixed now?

closed 2020-05-26 by gfwilliams