espruino/BangleApps #157: Error installing app to watch

paulcockrell, 2020-03-28:
Installing an app I am developing fails to install from forked app-store.

The app is currently 550 loc  @16Kb, and set to grow much larger. When installing from the development IDE into RAM/File, it installs and runs just fine. But from the forked app-store I get the following error:

```
...
<BLE> Sending "ng\\\"});\\n}\\n\\nin"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "it();\");;Bluetoo"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "th.println(\"OK\")"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Received "Uncaught SyntaxError"
puck.js:362 <BLE> Received ": Got UNFINISHED STR"
puck.js:362 <BLE> Received "ING expected EOF\r\n a"
puck.js:362 <BLE> Received "t line 1 col 45\r\n..."
puck.js:362 <BLE> Received ").write(\"astrocalc.a"
puck.js:362 <BLE> Received "pp.js\",\"/*\\n Origina"
puck.js:362 <BLE> Received "l source license:\\n\\"
puck.js:362 <BLE> Received "...\r\n               "
puck.js:362 <BLE> Received "               ^\r\nEx"
puck.js:362 <BLE> Received "ecution Interrupted\r"
puck.js:362 <BLE> Received "\nNew interpreter err"
puck.js:362 <BLE> Received "or: LOW_MEMORY,MEMOR"
puck.js:362 <BLE> Received "Y\r\n>"
```

The UI reports an `UNFINISHED STRING`, but that console error looks like its a low memory err?

I currently have 3,535,216 storage free according the file-manager app on the watch.

brainfart-be, 2020-03-29:
Hi ! 
I got that issue today, i managed to fix it by reducing my file size to less than 10kb.
I saw that 10kb is the max size limit.
I don't know if there is workaround, i'm not familiar enough with the filesystem of the watch, i tried to put one of the library i use into a separated file without success.

Hope it help !

paulcockrell, 2020-03-29:
Hi @brainfart-be 

Thanks for the tip on the max file size. I have ran my code through a compressor (https://skalman.github.io/UglifyJS-online/) which sorts the problem for now, but means I have to maintain two repos, one with original source, and this one with a compressed version.

Hopefully in the future either the IDE compresses files before it sends them up, the max file size is increased, or as you said you can split the app across multiple limited sized files.

brainfart-be, 2020-03-29:
`function evalFile(filename){
  var code = require('Storage').read(filename);
  eval(code);
  delete(code);
}`

Using this javascript snippet, you can import some code from another file (don't forget to register it in apps.json) 
Not sure it's the correct thing to do, but it seems to work

gfwilliams, 2020-03-30:
You can use `require('filename.js')` too I believe.

This is something that should be fixed in the app loader - it can split larger files into multiple Storage.write commands: http://www.espruino.com/Reference#l_Storage_write

commit 04f8cbcd (2020-03-31): write code in chunks, in case it is too big to fit in RAM (fix #157)
closed 2020-03-31 by gfwilliams