espruino/BangleApps #182: Settings: widgets

rigrig, 2020-04-01:
I was thinking that maybe it would be nice if we could add widget settings to the settings app, instead of having separate "apps" just to configure a widget?

Something like
* A widget can set `WIDGETS["mywidget"].settings` to a function to handle settings
* The settings app gets a new submenu "Widgets", containing (only) all widgets that have a `settings` property
* Ideally widgets also set `name`, which is used as menu entry
    - If not set, fall back to the `WIDGETS` key? So it would just show as "mywidget"
    - Or should `.settings` contain both a name and a callback?
    - Maybe we could get the name from `apps.json`? Seems like not all widgets use their  `id` as `WIDGETS` key though

So you'd have something like this:
```js
  function settings(back) {
    const mySettingsMenu = {
        '': { 'title': 'My Widget' },
        'Sound': {
            value: false,
            onchange: s => {s ? Bangle.beep() : false} // TODO: save new value
        },
        '< Back': back // provided by caller (settings app)
    };
    E.showMenu(mySettingsMenu)
  }
  WIDGETS["mywidget"]={
    /* other properties go here */
    name:"My widget",
    settings:settings,
  };
```

gfwilliams, 2020-04-02:
That sounds like a good idea. Maybe this could be more general though?

For instance if there was an `app.settings.json` it could be shown and loaded up. It's not just widgets that could benefit from this.

In terms of implementation there's `appname.info` that's auto-generated by the app loader, so we could add an entry in there if a settings file exists on upload.

rigrig, 2020-04-02:
That does seem a better solution, I'll give implementing something like that a try.

gfwilliams, 2020-04-02:
Nice - thanks! For detecting settings and setting the relevant bit in the app.info, just copy/paste this:

https://github.com/espruino/BangleApps/blob/master/js/appinfo.js#L61,L62

Then finding a list of apps with settings would look a lot like:

```
var apps = s.list(/\.info$/).map(app=>s.readJSON(app,1)).filter(app=>app&&app.settings);
```

PR #194 (2020-04-03): Settings: Add support for app/widget settings
Fixes #182

Apps and widgets can add a `<appid>.settings.js` file which can be opened from the "App/widget settings" submenu.
This file should define a single function to configure the app/widget.
The function is passed a `back` argument, which can be used to return to the settings menu by calling `back()`.

Example `<appid>.settings.js`:
```
(function(back) {
  const mySettingsMenu = {
    '': { 'title': 'My Widget' },
    '< Back': back, // go back to settings menu
    'Sound': {
      value: false,
      format: s => s ? 'on' : 'off',
      onch [... cut]
closed 2020-04-03 by gfwilliams