espruino/BangleApps #186: Settings missing brightness

marcocastignoli, 2020-04-02:
In the settings app there is no brightness selector. For now I made an app for managing brightness:

I created a widget that doesn't draw anything but set the brightness taking the value from a value save in the storage by my new brightness app.
Then I had to edit the launch.app.js because widgets are not loaded there.

But I think it would be easier to have the brightness selector inside the settings app :)

gfwilliams, 2020-04-06:
Reopen because PR won't auto-load settings at boot. 

It'll need something in `settings.boot.js` as well I imagine...

Didgeridoohan, 2020-04-08:
From what I see it doesn't even keep the setting after you've left the Settings app. The LCD brigthness changes as you change the setting, but as soon as you go back out it goes back to full brightness. The setting value is saved in the menu, but does nothing (until you go back and change the value again, but as previously the result is temporary).

It might also be a good idea to limit the min value to 0.1, so that it isn't possible to accidentally turn off the LCD completely.

gfwilliams, 2020-04-08:
Thanks, yes - that's why this issue is still left open...

closed 2020-04-08 by gfwilliams

Didgeridoohan, 2020-04-08:
Didn't first realise that I was actually reporting the exact same issue as you where... Now I get you.

I'm still trying to learn how everything is set up. Works fine now.

PR #234 (2020-04-06): Adds LCD brightness setting
Fixes #186
commit 50e61015 (2020-04-08): Settings: Make LCD brightness work after leaving settings (fix #186)