espruino/BangleApps #201: Help/Readme pages

gfwilliams, 2020-04-03:
As apps get more complicated I think it'd make sense to have some docs on how use use them.

Maybe an optional `README.md` file in the app directory, and then if there was ``"readme":"README.md"` in `apps.json` we could display a little help icon?

MaBecker, 2020-04-03:
Sounds like a nice way to add more information about the app or widget.

Like how to use it or special technic that was used to design and programming it and  some pictures would be nice too.

The App Loader is really cool - I like it!

gfwilliams, 2020-04-03:
Yes, I've been wondering about screenshots. They would be super-helpful, it's just a shame you can't read back from the display unless the app runs in the emulator.

MaBecker, 2020-04-04:
> it's just a shame you can't read back from the display 

What about overwriting setPixel via javascript to tee (Unix tee)  to a storage file and than use the WebIDE to pull this off the device - just for testing, even if this will be slow - if possible.

MaBecker, 2020-04-04:
Just saw someone adding a gif file, so gifs would be cool too.

gfwilliams, 2020-04-04:
Overriding setPixel would be cool (insanely slow though!). I guess the easy (and probably more friendly) option is to encourage apps to be written such that they would run in the emulator?

gfwilliams, 2020-04-07:
Closed thanks to @paulcockrell!

closed 2020-04-07 by gfwilliams