espruino/BangleApps #210: Bangle.js Launcher improvements

ps-igel, 2020-04-03:
Wearing the Bangle for a week now, these issues pop up frequently:

1. BTN2 often accidentally presses and it ends up showing the launcher
2. while in launcher widget (i.e. GadgetBridge) don't work because they are probably not loaded
3. returning to the clock using BTN3 takes a bit too long for my taste

suggestions:
- introduce setting to choose BTN1 for showing launcher
- make launcher return to default clock when LCD times out
- reduce the amount of time BTN3 needs to return to clock (maybe only a tiny bit)

gfwilliams, 2020-04-03:
Hi,

It sounds like the main issue here is BTN2 accidentally opening the launcher? One option would be to disable waking the watch up on BTN2 press, so you had to press some other button to have the watch displaying first, (*then* press BTN2?)

That's actually something that can be done in code right now: `Bangle.setOptions({wakeOnBTN2:false})`

So realistically it's just the ability to change that from setting that would be needed?

Right now the clocks themselves are responsible for using BTN2 for the launcher, so you could make your own clock app that used BTN1.

> make launcher return to default clock when LCD times out

That sounds like a great idea.

> reduce the amount of time BTN3 needs to return to clock (maybe only a tiny bit)

Sadly it's that long precisely because someone else complained it wasn't long enough! Can't please everyone :)

I guess it could be configurable though...

ps-igel, 2020-04-03:
> One option would be to disable waking the watch up on BTN2 press

I already have, still happens though. Probably when turning the wrist in combination with tight jacket arms.

> Right now the clocks themselves are responsible for using BTN2 for the launcher, so you could make your own clock app that used BTN1.

Right, didn't think on that. That'll be enough for the moment I guess.

> I guess it could be configurable though...

That'll be fine. No hurry though.

Thanks!

brainfart-be, 2020-04-03:
@ps-igel
If you want to give it a try, i developed another laucher - Touch launcher 
( https://espruino.github.io/BangleApps/#tool ) 
If you like it, i can easily add the feature to close the launcher when the display turn off.

ps-igel, 2020-04-04:
@brainfart-be 
I like the launcher. Looking forward for the new feature.

brainfart-be, 2020-04-04:
Thank's, it has been merged today. (v0.3)
Due to a limitation (probably on the firmware), when the LCD turn off and the launcher quit, the lcd turn on again on the clock, but at least it quit the launcher

ps-igel, 2020-04-04:
@brainfart-be 
Just tested it, looking good thanks!
Another improvement idea: can you make the menu revolving? So when you hit the last app, start from the first app again and vice versa?

brainfart-be, 2020-04-04:
I'll think about it, it break the "Back" page (when you swipe to the left ), this was needed by me because i don't like to have to click a long press on the button. 
But now that the launcher close itself, the "back" page may be useless.
I'll let you know :D

gfwilliams, 2020-04-04:
While we're making suggestions, if it were possible to animate it'd be amazing :)

It may be that to do that nicely you'd need to use the 120x120 or 80x80 mode though.

brainfart-be, 2020-04-04:
I will surely be bored tonight, 
I will do both .
Not sure about how to animate it yet, but that's where the fun begin !

ps-igel, 2020-04-08:
@brainfart-be 
Hey, I just upgrade to v0.04, hate to say this BUT: the animation really slows down the cycling through the apps. Way too much for my taste anyways. Any chance of speeding that up or make it optional. Maybe it only animate with swipe and does it without when touching instead?

gfwilliams, 2020-04-08:
Just had a quick look at the code, and it seems like you *might* be rendering all icons, regardless of whether they are onscreen or not?

Also, since you only have two icons onscreen at a time, you could load those into memory with `Storage.read(app.icon)` before doing the animation, and it'd really speed stuff up.

brainfart-be, 2020-04-08:
Indeed, i used to have an if statement to ignore what was not on the screen, i guess i removed it and forget to put it back , that's an easy fix.

For the store.read(app.icon), it make perfect sense to read the two icon before the animation and not at each frame. 

Thanks for the tips!

Didgeridoohan, 2020-04-09:
@brainfart-be I do see a slight improvement in the animation rendering in v0.05, but would appreciate an option to disable animation altogether.

ps-igel, 2020-04-09:
> @brainfart-be ... but would appreciate an option to disable animation altogether.

I'd prefer this too. The animation is still too slow for a good user experience.

brainfart-be, 2020-04-09:
I'll take a look at how i can integrate the settings in the app.
I think the same about the animation.
Do you think i need to default with or without animation?

brainfart-be, 2020-04-09:
Ok! I did a complete rewrite of the app in 80x80 lcd mode.
I manage to get a render time under 100ms (~80ms), before it was around 150ms.
I also added settings: (animation on/off, frame rate (number of frame to render the animation), debug on/off)
Before i do another PR, i let you test the v0.06 from my dev app store: https://bangle.dimitri-gigot.com/#tool

Didgeridoohan, 2020-04-10:
@brainfart-be Much better performance! I will probably keep the animation enabled now...

Two things that I noted:
At least for me, it would be preferable to have the app settings in the Settings app.
The Settings app shortcut in the launcher takes me to the app settings rather than the watch settings. Oops...

brainfart-be, 2020-04-10:
Thanks @Didgeridoohan , I'm happy you like it

I'm not sure to understand what you mean bout the settings.
There is two type of "page", some with an icon, those are other apps on the watch, and there are two screens with just some text "Exit" and "Settings", those are part of the launcher.
I guess you expected the "Settings" page to open the "Watch settings app" instead of the launcher settings right?
I need to find a way  do distinguish the "Exit" and "Settings" page from the rest 

Can't wait to make the perfect Launcher haha
Thanks for the inputs! really appreciate it!

ps-igel, 2020-04-10:
Ok, its faster now but the 80x80 resolution is not my thing. I really liked the version from around when I started this thread. :-)
Anyhow, I noticed too that there are two pages: Settings without an image and settings with an image. Both take me to the settings of the app rather than the settings of the watch. 
Maybe you can integrate the settings of the app in the app/widgets settings part of the watch settings using toucher.settings.js.

Didgeridoohan, 2020-04-10:
@brainfart-be I got that the "page" with just the "Settings" text is for launcher settings. This is what I think would be useful to instead have in the main settings app, under the "App/widget settings" submenu. See here for reference:
https://github.com/espruino/BangleApps/pull/194

The bug I raised was that when I wanted to open the watch settings (the "page" with a cogwheel icon) it took me to the same launcher settings rather then opening the watch settings.

brainfart-be, 2020-04-10:
Oh ok got it!
@Didgeridoohan i'll take a look at the reference you send there ;)
@ps-igel I was quite sure i didn't did it right :D i didn't thaught about loading a different js file and run it like an app instead of my weird openSettings function.

Oh and yeah.. sorry for the 80x80 resolution, i really liked the animation so i wanted to make it work.
I will try to see if i can make the lcd mode dynamic to have a "high res" and a "low res" 
high res + no animation should work like the version you loved
for low res i had to remove stuff like version number and type of app, but this can be added again on the high res version

Like always, that's what i'm going to spend my quarantine evening to do !

Again thanks !
It's the first time i do as much as opensource and it really make me happy :p

brainfart-be, 2020-04-10:
The code will need to be cleaned and comment before i do the PR but i implement all the stuff you asked, i hope this time i did it right :p 
Enjoy (and let me know) !
https://bangle.dimitri-gigot.com/#tool

ps-igel, 2020-04-10:
Strange, when trying to upload I always get this halfway through:

`<BLE> Sending "TEP;\\n\\n  if(STA"`
`<BLE> Sent`
`<BLE> Sending "TE.offset >= STA"`
`<BLE> XOFF received => pause upload`
`<BLE> Received ""`
`<BLE> Sent`

and the upload gets stuck. Not sure what the reason is, never seens that before.

Didgeridoohan, 2020-04-10:
I tested earlier and opening the watch settings now works as expected. But there are no Touch Launcher settings under the "App/widget settings" submenu.

brainfart-be, 2020-04-10:
@Didgeridoohan  It should be displayed as "Toucher" at least that what i got on mine.
@ps-igel  That's weird :s i cannot help you on that one

Didgeridoohan, 2020-04-10:
@brainfart-be Hmm... Nope, it wasn't in there at all (my other apps that supports it, Battery widget with % and Gadgetbridge was showing just fine). I guess I'll check again tomorrow, now it's way past my bedtime...

ps-igel, 2020-04-11:
@Didgeridoohan The toucher setting is visible for me.

@brainfart-be I like the combination of high-res and no animation! Thanks! :)

brainfart-be, 2020-04-11:
I will clean that messy code and do a PR tomorrow.
Glad you like it :D

Didgeridoohan, 2020-04-11:
Oh right... Forgot to report back on that. Had to reinstall v0.06, but after that the Toucher settings showed up as expected.

gfwilliams, 2020-04-23:
Looks like this is sorted now - you can turn BTN2 off in settings, and there's a neat new launcher as well :)

closed 2020-04-23 by gfwilliams