espruino/BangleApps #216: Removing an app should look at `app.info` `files`

rigrig, 2020-04-03:
When removing/updating an outdated app, the list of files on the device might no longer match the list in `apps.json`
See for example #215, where `gbridge.app.js` was changed to `gbridge.settings.js`.

I think removing an app should get `app.info` from the device to determine which files to remove.

gfwilliams, 2020-04-04:
Good point - I think that data should already be in `appsInstalled`.

rigrig, 2020-04-04:
I just thought of some complications:

* If you just opened the page, `appsInstalled` stays empty until you refresh from My Apps, but you can just install apps, overwriting whatever was on the device.  
* What about settings files like `app.settings.json`? 
  - Fresh install: might include a default `settings.json` file?[1]
  - Update: we don't want to overwrite settings already on the watch
  - Uninstall: settings file should also be cleaned up
    (I think "real" data like gps/heartrate recordings should be kept though)

[1] But you need to put defaults in the code anyway, or adding more settings in a newer version becomes problematic.

So I think it would need to do something like this:
1. Add a way to indicate apps contain a settings file, something like this:
  ```json
  "storage": [
    {"name":"app.settings.json","type":"storage"},
  ```
  Files with `type=="storage"` are allowed to have no `url/contents/evaluate`, in which case nothing is actually uploaded to the device.

##### Install: 
  * If `app.info` is already on the device, perform update instead (confirm first?)
  * If a file has `type:"storage"`, add it to the `app.info` `files` list, even if it has no contents (then don't actually upload anything)
##### Update:
  * Get `app.info` from device
  * Check `app.info` files: delete all files not listed in `app.storage` (others get overwritten anyway)
  * If a file has `type:"storage"`, do not overwrite if it already exists on the device
##### Remove:
  * Get `app.info` from device, delete all listed files.

gfwilliams, 2020-04-06:
Thanks -  what you propose above looks good, but I'm a bit iffy about the PR itself. Creating `new RegExp` and then serialising it and sending it over to run on Bangle.js seems a bit dodgy... Also it's unquoted, so `a.foo.js` will match the file `abfoocjs`.

I'd also really prefer to keep things like `listFiles` out of the upload process if at all possible. I'd imagine that given the settings files are tiny, it might be worth actually just changing the upload text to something more like: `if (require("Storage").read(filename)===undefined) require("Storage").write(...)`

My concern is that I'm not sure it actually solves the problem completely. It keeps the settings, but for heart rate, pedometer, gps recorder, stopwatch that create files it won't actually be able to remove those as they tend to have different names (eg with the date in them). I think we might need a separate list of regexp for data files rather than just a tag on the stuff in `storage`.

If we just ensured that whenever you click `upload`, the app list (appsInstalled) was first grabbed from the device I think that would make life a lot simpler. There's then no need to get `app.info` from the device on update because it's already in appsInstalled.

rigrig, 2020-04-06:
Created a cleaner PR for just looking at `appid.info`: #242

I'll open a new issue for keeping data/storage/settings files, as I think it'll benefit from some more discussion without entangling it with this.

PR #242 (2020-04-07): Improve update process
Fix #216
Cleaner PR than #235: just improves the update process without worrying about special settings/data files.

- Always fill `appsInstalled` before uploading
- Install: if `appid.info` already on device, update instead
- Update: get files to remove from `appid.info`, only remove files which
          won't be overwritten anyway
- Remove: get files to remove from `appid.info`
closed 2020-04-07 by gfwilliams