espruino/BangleApps #219: What about a App to customise my Banglejs?

MaBecker, 2020-04-04:
A app with options and value inputs for Bangle.setOptions() plus a free text input for optional stuff.

* Upload: store on device only option different to default and create cookie
* Default: rest values on app screen to default
* ...

Allows non developers to change default setting via app.

gfwilliams, 2020-04-04:
IMO the setOptions stuff should just be in Settings by default. There is an issue open for that...

But a custom app that allowed JS to be pasted into a boot.js would be cool, and would handily avoid your app removal issue if combined with favourite apps

MaBecker, 2020-04-04:
Ok, so let me try this.

gfwilliams, 2020-04-04:
Great! It'd be cool to use a syntax highlighted editor like CodeMirror, and maybe start off with some example code commented out that folks could uncomment and change?

MaBecker, 2020-04-06:
Have you planed to extend js/utils.js  by setCookie() and get getCookie()?

MaBecker, 2020-04-06:
```
function setCookie(cname, cvalue, exdays) {
  var d = new Date();
  d.setTime(d.getTime() + (exdays*24*60*60*1000));
  var expires = "expires="+ d.toUTCString();
  document.cookie = cname + "=" + cvalue + ";" + expires + ";path=/";
}

function getCookie(cname) {
  var name = cname + "=";
  var decodedCookie = decodeURIComponent(document.cookie);
  var ca = decodedCookie.split(';');
  for(var i = 0; i <ca.length; i++) {
    var c = ca[i];
    while (c.charAt(0) == ' ') {
      c = c.substring(1);
    }
    if (c.indexOf(name) == 0) {
      return c.substring(name.length, c.length);
    }
  }
  return "";
}
```

Does that save them local?

MaBecker, 2020-04-06:
```
function checkCookie() {
  var username = getCookie("username");
  if (username != "") {
   alert("Welcome again " + username);
  } else {
    username = prompt("Please enter your name:", "");
    if (username != "" && username != null) {
      setCookie("username", username, 365);
    }
  }
}
```

MaBecker, 2020-04-06:
Is storing CodeMirror in  css/codemirror and lib/codemirror like this ok ?

[image]
[image]

gfwilliams, 2020-04-06:
Yes, that'd be fine... Or you could just use a CDN - at least for now. https://cdnjs.com/libraries/codemirror

I guess that might be preferable... As this is just one app, it's a lot of extra stuff to add.

In terms of setCookie/etc, maybe just use localStorage? https://github.com/espruino/BangleApps/blob/e936aecf8484f2a5321f09f1a7eabebea70ea254/apps/grocery/grocery.html#L79

Just came across it in that PR and it looks really nice and tidy :)

MaBecker, 2020-04-06:
Cool, cdnjs is much easier.

Not sure about  `localStorage.setItem` technic, as it  will be lost after a `Remove all Apps`.

gfwilliams, 2020-04-06:
`localStorage` is local to the Browser, not the Bangle :)

MaBecker, 2020-04-06:
add to apps.json. Sortorder just set for easy testing.

```
  { "id": "cbc",
    "name": "Custom Boot Code ",
    "icon": "cbc.png",
    "version":"0.01",
    "description": "Add code you want to run startup up",
    "tags": "tool,system",
    "type": "bootloader",
    "custom":"cbc.html",
    "storage": [
       {"name":"custom.boot.js"}
    ],
    "sortorder" : -99
  }
```

cbc.html

```
<html>
<head>
    <link rel="stylesheet" href="../../css/spectre.min.css">
    <link rel="stylesheet" href="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.32.0/codemirror.min.css" />
    <style>
    .CodeMirror {
        border: 1px solid #eee;
        height: auto;
    }
    </style>
</head>
<body>
    <script src="../../lib/customize.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.32.0/codemirror.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/mode/javascript/javascript.min.js"></script>
    <p>Type your javascript code here</p>
    <p><textarea id="custom-js" rows="4" _cols="70"></textarea></p>
    <p>Then click <button id="upload" class="btn btn-primary">Upload</button></p>
    <script>
        var appid = "cbc";
        var item = "custom.boot.js";
        var id = "custom-js";
        var sample = "//Bangle.setOptions({wakeOnBTN2:false});";

        if (localStorage.getItem(item) === null) {
            localStorage.setItem(item, sample);
        }

        document.getElementById(id).value = localStorage.getItem(item);

        var editor = CodeMirror.fromTextArea(document.getElementById(id), {
            lineNumbers: true,
            mode: "javascript",
            utters: ["CodeMirror-lint-markers"],
            lint: true,
        });

        document.getElementById("upload").addEventListener("click", function() {
            var localeModule = document.getElementById(id).value;
            localStorage.setItem(item, localeModule);
            sendCustomizedApp({
                storage: [
                    { name: appid, content: localeModule }
                ]
            });
        });
    </script>
</body>
</html>
```

* codemirror needs some settings to fit on custom page

[image]

[image]

* syntax check would be nice
* save code to `cbs` and not to `custom.boot.js`

[image]

gfwilliams, 2020-04-07:
Looks great - Pull request! Could I suggest:

```
var code = localStorage.getItem(item);
if (code===null) {
  code = `
// uncomment/modify lines to customise your Bangle.js
//Bangle.setOptions({wakeOnBTN1:false})
//..more JS code you might want to run?
`;
}
document.getElementById(id).value = code;
```

MaBecker, 2020-04-07:
Well not happy, because of this 

> saves code to cbs and not to custom.boot.js

any hints how to save to `custom.boot.js`

MaBecker, 2020-04-07:
content of cbc
```
{"id":"cbc","name":"Custom Boot Code ","type":"bootloader","sortorder":-99,"version":"0.01","files":"cbc.info,cbc"}

```

MaBecker, 2020-04-07:
not cbc, but cbc.info

cbc contains the javascript code

gfwilliams, 2020-04-07:
> any hints how to save to custom.boot.js ?

Literally just:

```
sendCustomizedApp({
                storage: [
                    { name: "custom.boot.js", content: localeModule }
                ]
            });
```

Although maybe you could name the whole app `custom` rather than `cbc`? It makes things cleaner to have them all share the same first bit of the name.

MaBecker, 2020-04-07:
cbc was just the working title ;-)

I thought the  filename is pick from the apps.json, but using the filename work.

OK, so let's push this into a branch and create a pull request.

MaBecker, 2020-04-07:
Don't know how to get this fixed:

Error: window.JSHINT not defined, CodeMirror JavaScript linting cannot run. VM2807 javascript-lint.min.js:1 

[image]

```
<html>

<head>
    <link rel="stylesheet" href="../../css/spectre.min.css">
    <link rel="stylesheet" href="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.32.0/codemirror.min.css">
    <link rel="stylesheet" href="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/addon/lint/lint.min.css">
</head>

<body>
    <script src="../../lib/customize.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.32.0/codemirror.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/mode/javascript/javascript.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/addon/lint/lint.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/addon/lint/javascript-lint.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/codemirror/5.52.2/addon/hint/javascript-hint.min.js"></script>
    <p>Type your javascript code here</p>
    <p><textarea id="custom-js" rows="4" _cols="70"></textarea></p>
    <p>Then click <button id="upload" class="btn btn-primary">Upload</button></p>
    <script>
    var item = "custom.boot.js";
    var id = "custom-js";
    var sample = "//Bangle.setOptions({wakeOnBTN2:false});";
    var localeModule = null;
    var editor = {};

    if (localStorage.getItem(item) === null) {
        localStorage.setItem(item, sample);
    }

    document.getElementById(id).value = localStorage.getItem(item);

    // The code editor
    var lintFlags = {
      esversion   : 6,    // Enable ES6 for literals, arrow fns, binary
      evil        : true, // don't warn on use of strings in setInterval
      laxbreak    : true,  // don't warn about newlines in expressions
      laxcomma    : true  // don't warn about commas at the start of the line
    };
    editor = CodeMirror.fromTextArea(document.getElementById(id), {
        width: "100%",
        height: "100%",
        lineNumbers: true,
        matchBrackets: true,
        mode: { name: "javascript", globalVars: false },
        lineWrapping: true,
        showTrailingSpace: true,
        lint: lintFlags,
        gutters: ["CodeMirror-linenumbers", "CodeMirror-lint-markers"]
    });

    document.getElementById("upload").addEventListener("click", function() {
        localeModule = editor.getValue()
        localStorage.setItem(item, localeModule);
        sendCustomizedApp({
            storage: [{ name: item, content: localeModule }]
        });
    });
    </script>
</body>

</html>
```

MaBecker, 2020-04-07:
```
  { "id": "custom",
    "name": "Custom Boot Code ",
    "icon": "custom.png",
    "version":"0.01",
    "description": "Add code you want to run at boot time",
    "tags": "tool,system",
    "type": "bootloader",
    "custom":"custom.html",
    "storage": [
       {"name":"custom"}
    ],
    "sortorder" : -99
  }
```

gfwilliams, 2020-04-08:
> window.JSHINT not defined

You need to include a `script` tag for `jshint` - the codemirror stuff provides everything to wrap around it, but not jshint itself.

closed 2020-04-08 by MaBecker