espruino/BangleApps #223: Issue drawing when lcdMode is set to 80x80 or 120x120

brainfart-be, 2020-04-04:
I wanted to test different mode for the display. 
I don't manage to make this code draw a white square (either in emulator and on the device)
or even clear the screen
when the "Bangle.setLCDMode()" is commented, it work like expected.

`// Initialise display
Bangle.setLCDMode("120x120");

// Store screen dimensions
let W = g.getWidth();
let H = g.getHeight();

function draw(){
g.clear();
  console.log(W, H);
  g.setColor(1,1,1);
  g.fillRect(10,10, 110,110);
}

setInterval(draw, 500);`

Maybe i'm doing something wrong

paulcockrell, 2020-04-04:
Do you need to add `g.flip()` to invoke the draw

brainfart-be, 2020-04-04:
Ooooh thank's !
I saw that g.flip function in a lot of code i read but didn't imagine that could be it (time to check the doc to understand what's happening behind!

closed 2020-04-04 by brainfart-be