espruino/BangleApps #238: Apps Reviews?

gfwilliams, 2020-04-06:
Is there a service we can hook into for reviews (and feedback?) without having to implement something? Eg, being able to keep the website static.

I imagine soon there will be multiple apps that do the same thing, and for new users it'd be good to have a way for them to see what the most useful app is or get some tips and tricks.

It may be the best solution is just to have a Wiki page on BangleApps for each app?

Purple-Tentacle, 2020-04-07:
I would really like this to be part of the app loader.
GitHub cannot display dynamic pages?

Fabio286, 2020-04-19:
Maybe we can create an api server and host it somewhere.
In this case it would be perfect for a static web application.

gfwilliams, 2020-04-20:
Yes, I think that's probably going to be the way forward.

I'm using Auth0 on the forum and as part of the KickStarter I said I'd add some sign-in based features to the Web IDE. I think it'd probably make sense for me to store reviews/ratings using the same sign-in.

Fabio286, 2020-04-20:
If you think it's a good idea i can try to develop a REST API server integrating Auth0 authentication via JWT, and store user preferences and reviews on database.
I would opt for a full JS approach using Node.js with Express framework and [Knex](https://github.com/knex/knex) as query builder.
What do you think?

gfwilliams, 2020-04-21:
Thanks - sounds like a good idea tech-wise.

If it's ok to wait a bit it's something I'd quite like to do as it'll probably end up running on my server, and I'd like to do it in such a way that I can use it for the Web IDE stuff too.

Fabio286, 2020-04-21:
Nice!
I hope Bangle universe grows, because i believe there is a bunch of amazing ideas starting from the watch itself to end with the fully web based IDE and app store.
If i can, i'll gladly help.

gfwilliams, 2024-05-02:
I don't think there's enough interest in this - we now have sort by favourites and installs an which allows apps people like to get surfaced without worrying about out of date comments, moderation, and people being mean to each other

closed 2024-05-02 by gfwilliams