espruino/BangleApps #239: Low on battery warning

veelco, 2020-04-06:
Would be awesome to have a warning if my battery is running low. It just shuts down right now. I'm using the Tiny Battery Widget currently which is some sort of an indication, but I like to have a pop-up message/vibration at 5% so I can recharge my watch.

Purple-Tentacle, 2020-04-07:
@gfwilliams 
That should be part of the discussed best practices documentation.
Every battery widget should use that imo.

gfwilliams, 2020-04-07:
What do you mean? Every widget should buzz for low battery?

Purple-Tentacle, 2020-04-08:
Every battery widget should.
Or not. 
But all should behave in the same way imo.

MaBecker, 2020-04-08:
There is a folder for snippets like the one you are talking about

https://github.com/espruino/BangleApps/tree/master/testing

To be honest: All apps and widgets are designed by individual persons in the way they like them to look and work. I suggest make a copy of tiny battery widget and change the new widget the way you like it.

 **Building Apps and Widgets the way YOU like them is the one of the biggest benefit you get with Bangle.js**

Purple-Tentacle, 2020-04-08:
> Building Apps and Widgets the way YOU like them is the one of the biggest benefit you get with Bangle.js
Agreed. And I am already doing such things in creating stuff I want and changing stuff the way I want it.
I am just thinking of a streamlined experience with all apps in the "official" app loader.
And that would mean that some things should be consistent in all apps.

In the end, I am not the one making the decision, I am just proposing.

gfwilliams, 2020-04-08:
Thanks - however in this case I'm with @MaBecker. I feel like the whole point of this is to put as few restrictions on what can be done as possible so the watch can be customised as much as needed.

I can imagine what might happen in a lot of cases is we end up with one 'basic' version of an app/widget, and one 'kitchen sink' one that has everything and a bunch of settings?

I'm not 100% sold on the low battery warning - for instance if it happens overnight it's just going to annoy a bunch of people. I could totally imagine something that catches your attention when you look at the watch though, like a flashing red X?

Purple-Tentacle, 2020-04-08:
>Thanks - however in this case I'm with @MaBecker. I feel like the whole point of this is to put as few restrictions on what can be done as possible so the watch can be customised as much as needed.

Ok, fair enough :)

>I'm not 100% sold on the low battery warning - for instance if it happens overnight it's just going to annoy a bunch of people

I am used to set my phone to quiet before I go to sleep. The same should be done with a watch in this case. Otherwise other things may go of as well. Notifications from GB, for example.

Which brings me to another idea.
What about a global scheduled silenced mode where beep and vibrate are turned off automatically and then turned back to their original setting again?

gfwilliams, 2020-04-08:
> What about a global scheduled silenced mode

Sounds like a plan. It could easily be made into a widget (or something that runs in xyz.boot.js).

MaBecker, 2020-04-08:
Let me add global `scheduled dark mode` which  disable all lcd turn on too , except BTN1 or so.

v1nc, 2020-06-12:
> Would be awesome to have a warning if my battery is running low. It just shuts down right now. I'm using the Tiny Battery Widget currently which is some sort of an indication, but I like to have a pop-up message/vibration at 5% so I can recharge my watch.

You could create an visually empty widget, that uses the [notification handler](https://github.com/espruino/BangleApps/tree/master/apps/notify) and only display something if the battery is low. This should not interference with other battery widgets.

gfwilliams, 2020-06-12:
Yep - with the new notifications this should be pretty easy

rigrig, 2020-07-20:
It was :-)

jjok, 2020-08-12:
Looks like this should have been closed by #531.

rigrig, 2020-08-12:
I was planning to split the silent mode to a different issue, but forgot about it. Done now: https://github.com/espruino/BangleApps/issues/544

closed 2020-08-17 by gfwilliams