espruino/BangleApps #241: Gadgetbridge & BangleApps: Timezone Issue?

fungus75, 2020-04-06:
I'm using CET+1 but since last week CEST+2.
And this generates some troubles:

Round about every 30 minutes, the Watch adjusts itself back to CET+1 (looks like), because i.E. it shows 15:30 when it is 16:30.
Long Press on BTN3 restarts the Watchface and it's in the correct time again.
I have adjusted in the Settings Locale=>Timezone to 2.

I guess there must have been something with Gadgetbridge, because when turning off Blutooth on the Phone, the Bangle keeps the correct time.
It worked perfect on CET+1, the problem occured when changing the timezones.

Phone should be set correct, because it shows the correct time.
I'm recendly updated everything on Bangle to the newest available versions and using Gadgetbridge V 0.43.0

Didgeridoohan, 2020-04-06:
That is indeed a Gadgetbridge issue combined with the fact that the Bangle can't yet deal properly with daylight savings. See discussion here:
http://forum.espruino.com/conversations/345273/

It should already be taken care of though, at least on the Gadgetbridge side of things. The 0.43.1 release should be here very soon judging by the activity on Codeberg:
https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/CHANGELOG.md

gfwilliams, 2020-04-07:
Yes - closing this. I'm pretty sure if you disable Gadgetbridge for now (or update it when the new release is available) it should be sorted.

closed 2020-04-07 by gfwilliams