espruino/BangleApps #249: Surfacing new apps

gfwilliams, 2020-04-08:
Should we have a way to show new apps? Right now it's getting much harder to see what's there if you're looking to try new stuff.

I guess the simplest would be to make a script that pulled the initial commit & last modified dates out of GitHub and added them to a new file. I could then run it as part of the process of publishing to banglejs.com/apps...

myxor, 2020-04-08:
I had the same idea. It would be great to have some kind of area on top of the app list which "advertises" new apps.

MaBecker, 2020-04-08:
Yes, some additional filters would be nice  like

- added on date
- popularity 

Add a download counter for each app to have a  popularity criteria 

Subscription list to get informed about updates and new entries

Purple-Tentacle, 2020-04-08:
And not only new apps, but updated ones as well.
Maybe some more views, where they are just sorted differently?

gfwilliams, 2020-05-12:
http://forum.espruino.com/conversations/347351/#comment15273702

gfwilliams, 2020-05-12:
Forgot to add, but fixed with https://github.com/espruino/BangleApps/commit/d03fb00f2ddfddbf09a9f9424308ff17131b9e5f - will be live tomorrow

closed 2020-05-12 by gfwilliams