espruino/BangleApps #250: Step counter goal

myxor, 2020-04-08:
Most of the fitness tracker watches have some configurable step goal for each day (e.g. 10.000 steps).
So the watch can show the progress of reaching the set goal and can notify when the daily goal is reached.

Wouldn't it be great to have something similar here for the people who use this as a sports tracker as well?

gfwilliams, 2020-04-08:
Yes, sounds good. Maybe it could be added to the pedometer widget?

myxor, 2020-04-08:
Good idea. Although I'm not sure how the information should fit in the limited space of the widget.

Maybe a clock with the step indicator would be a better idea?

Purple-Tentacle, 2020-04-08:
I am currently working on my own version of the pedometer widget, which filters out arm movements.
This includes an active state (or inactive state).
My idea was to show this with the color of the widget.
Green = Active
Red = Inactive.

Maybe the goal could be displayed with the fill state of the icon?

Purple-Tentacle, 2020-04-08:
I changed the display to one big number and a small number.
The big one shows 11k for 11834 steps.
The small one shows 11834.

The big one is red when inactive, green when active.

That improves readability quite a bit, unfortunately it does not leave any room for progress bar.

rigrig, 2020-04-08:
Didn't see the last few comments until I already made my pull request which displays the goal as a partial circle in the background. Maybe that could be adjusted to work with your changes?

closed 2020-04-09 by gfwilliams

myxor, 2020-04-09:
@rigrig 
Sorry to post here but after https://github.com/espruino/BangleApps/pull/260 got merged the widpedom widget seems to crash on my watch.
After i installed it and pressed BTN3 to reload interface the screen of the watch stays black and i can not open the main menu or anything else.
I can bring back the watch into a working mode only by removing the widpedom.
I am not sure how i can debug this on the watch itself as it seems not to happen on yours.

gfwilliams, 2020-04-09:
Just looked at this and it was obviously never tested :( Adding a fix now

rigrig, 2020-04-09:
Oops, sorry, that was stupid.
My editor made a bunch of autoformatting whitespace changes which I undid manually to not clutter up the commit. I must've broken it then and somehow failed to retest the final commit :-(

PR #260 (2020-04-09): widpedom: Add goal setting
Fix #250 
Draws progress as part of a circle
(default goal: 10.000 seems to be common-ish?)