espruino/BangleApps #261: app settings in main settings

ps-igel, 2020-04-08:
I think there is still an issue with using the myapp.settings.js to be added to the main settings page ([see this change](https://github.com/espruino/BangleApps/commit/39621bfae0c659e354840b31ff092e51ab6dca26?diff=unified))

I uploaded my own app using the myapp.settings.js file and the menu in it. But it does not show up in the settings. Right now I only see GadgetBridge in there.

One thing I noticed is, that gbridge.info file contains `"settings":"gbridge.settings.js"` whereas myapp.info does not contain any settings key-value-pair. But this seems to be the filter on which the settings app filters the apps.

The info file is auto-generated from apps.json right? Does it generate the settings key for widgets only and not for apps?

gfwilliams, 2020-04-09:
Do you have a link to your code?

`myapp.info` is automatically generated. In the code you linked you can see the line where it's added: https://github.com/espruino/BangleApps/commit/39621bfae0c659e354840b31ff092e51ab6dca26?diff=unified#diff-7564f118a462298ffcfe7959d3e8875cR64

It should create the pair regardless. Is it possible you've forked BangleApps but haven't updated it to the most recent version, which would have this in?

ps-igel, 2020-04-09:
> Is it possible you've forked BangleApps but haven't updated it to the most recent version, which would have this in?

I'm an *****. Most likely yes.

closed 2020-04-09 by ps-igel