espruino/BangleApps #263: Language pack breaks settings menu

paulcockrell, 2020-04-09:
# Issue
Language pack breaks settings menu labels. 

# Actual result
Some settings menu options display '[native code]' instead of expected label.

# Expected result
Correct settings menu labels displayed in English.

# Notes
Uninstall language app and menu back to normal

MaBecker, 2020-04-09:
Hi,

can you please add pictures to fix this - thanks.

paulcockrell, 2020-04-09:
@MaBecker 
[image]

gfwilliams, 2020-04-09:
Thanks - just fixed this with ea32401ba6fc944825c94dad69e27a5da61159c0

I'd spotted it once before but assumed I was using a broken dev firmware :)

closed 2020-04-09 by gfwilliams