espruino/BangleApps #267: Web UI: Getting app list failed, undefined

myxor, 2020-04-09:
In the dev Web UI on https://espruino.github.io/BangleApps/ i am getting the following error after app list is loaded:
 
> Getting app list failed, undefined

I can see this in the console:

> <BLE> Executing {"type":"eval","expr":"require(\"Storage\").list(/.info$/).map(f=>{var j=require(\"Storage\").readJSON(f,1)||{};j.id=f.slice(0,-5);return j})"} from queue
puck.js:362 <BLE> Sending "\u0010Bluetooth.print"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ln(JSON.stringif"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "y(require(\"Stora"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ge\").list(/.info"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "$/).map(f=>{var "
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "j=require(\"Stora"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ge\").readJSON(f,"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "1)||{};j.id=f.sl"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ice(0,-5);return"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending " j})))\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Received "Execution Interrupte"
puck.js:362 <BLE> Received "d\r\n>"
puck.js:362 <BLE> Unable to decode "Execution Interrupted\r", got SyntaxError: Unexpected token E in JSON at position 0

gfwilliams, 2020-04-09:
Do you have any other webpage open in the background that might also be connected?

Can you also try just restarting without loading code: https://www.espruino.com/Bangle.js#resetting-without-loading-any-code

The most likely possibility is you have some app/watch face that's running that is using up a bunch of CPU time and might be interfering with the upload process.

myxor, 2020-04-09:
Strange, after erasing one app via the app manager on the watch itself it starts working again.

ps-igel, 2020-04-09:
I have the exact same problem with own watch face. App loader shows

> <BLE> Executing {"type":"eval","expr":"require(\"Storage\").list(/.info$/).map(f=>{var j=require(\"Storage\").readJSON(f,1)||{};j.id=f.slice(0,-5);return j})"} from queue
>puck.js:362 <BLE> Sending "\u0010Bluetooth.print"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "ln(JSON.stringif"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "y(require(\"Stora"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "ge\").list(/.info"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "$/).map(f=>{var "
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "j=require(\"Stora"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "ge\").readJSON(f,"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "1)||{};j.id=f.sl"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending "ice(0,-5);return"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Sending " j})))\n"
>puck.js:362 <BLE> Sent
>puck.js:362 <BLE> Received "Execution Interrupte"
>puck.js:362 <BLE> Received "d\r\nNew interpreter e"
>puck.js:362 <BLE> Unable to decode "Execution Interrupted\r", got SyntaxError: Unexpected token >E in JSON at position 0
><BLE> **Received "rror: LOW_MEMORY,MEM"**
>puck.js:362 <BLE> Received "ORY\r\n>"
>puck.js:362 <BLE> Received "\r\n>"

Pointing to a low memory problem. When I connect via Web IDE and run `process.memory()` it shows

`={ free: 995, usage: 1105, total: 2100, history: 21,
  gc: 0, gctime: 6.28662109375, blocksize: 16, "stackEndAddress": 536923400, flash_start: 0,
  "flash_binary_end": 483740, "flash_code_start": 1073741824, flash_length: 524288 }`

If I switch to the alarms app and run process.memory again it shows:

`={ free: 1183, usage: 917, total: 2100, history: 0,
  gc: 0, gctime: 5.09643554687, blocksize: 16, "stackEndAddress": 536923400, flash_start: 0,
  "flash_binary_end": 483740, "flash_code_start": 1073741824, flash_length: 524288 }`

And then the app loader works. So there seems to be a certain amount of memory needed to load the apps list?

Purple-Tentacle, 2020-04-09:
Yeah, I had this when running low on memory.
I can help already to start another app which does not load the widgets, because the watch and the widgets together may take too much memory.

gfwilliams, 2020-04-09:
Maybe the app loader should just call 'reset()' even before loading the app list, to ensure there's enough memory.

There are ways to make it use less, but honestly someone will always have maxed everything out and had the issue

ps-igel, 2020-04-15:
>the app loader should just call 'reset()'

I tried this on my local app loader and that works nicely. Can we get this implemented? 

Maybe show a message on the Bangle to let people know whats happening? Not to confuse them why after connecting it shows the "reset screen".

gfwilliams, 2020-04-15:
Just fixed this with https://github.com/espruino/BangleApps/commit/a56a9792f15499ed3ee6f74a1fe6c70a5f22ab3d

It should be miles better now

closed 2020-04-15 by gfwilliams