espruino/BangleApps #268: GadgetBridge music commands not always working

Purple-Tentacle, 2020-04-09:
I extended the Gadgebridge widget.
Full code here:
https://github.com/Purple-Tentacle/BangleAppsDev/blob/master/apps/gbridge/widget.js

This is what I added to the end of the file:
`
  setWatch(function() { //BTN1
    if (state.music == "pause" || state.music == "stop") Bluetooth.println(JSON.stringify({t:"music", n:"play"}));
    else Bluetooth.println(JSON.stringify({t:"music", n:"pause"}));
  }, BTN1, {edge:"rising", debounce:50, repeat:true});
`

Connected to the IDE I can see that the play JSON is sent.
Connected to my phone, it seems that play is not always arriving on the phone.
Sometimes it needs 3 or 4 button presses to play the music.
After that it is almost always possible to toggle between play and pause with one click.

Is there something I am missing?

gfwilliams, 2020-04-09:
I don't think so - that all looks fine to me.

Maybe you could plug your phone in via USB and run `adb logcat` - you'll see all the log messages there including those that'd show if Gadgetbridge got the information

Purple-Tentacle, 2020-04-11:
I did not yet connect with USB, but some more information.
Sometimes I press the button when music is not playing. Nothing happens. Then 10-30 seconds later music starts to play.
If I press the button multiple times and nothing happens, sometimes all commands get stacked and are executed all at once in order. So music plays and pauses rapidly for 10 times, for example. 

That makes me think that the command are sent from Bangle.js, which I can see when connected to IDE. But somewhere on the way they get lost. 
Could be
- BLE from Bangle.js to phone
- in GB on the phone
- in GB on phone
- from GB on phone to music app

gfwilliams, 2020-04-14:
I'd need to know what you see on the phone. It's possible that somehow a lot of data gets sent to the phone (a stack trace from something broken?) and that delays the phone getting the command

closed 2021-02-02 by gfwilliams