espruino/BangleApps #276: Graphics createArrayBuffer().asImage()

rigrig, 2020-04-10:
I was looking into reducing flicker from redrawing widgets, and ran into some things happening I don't quite understand:
```js
const bpp = 24, setBuffer = true  // only works with these values
let b = Graphics.createArrayBuffer(10,10,bpp);
b.setColor(0xF800) // Red
b.fillCircle(5,5,4)
let i = b.asImage()
if (setBuffer) i.buffer = b.buffer
g.drawImage(i,10,10) // put the red circle on the display
```
1. Why does it need `setbuffer = true`? Isn't `b.asImage()` supposed to return a drawable image?
2. I expected that `bpp = 16` would work, because that should match the display. But unless I set `bpp = 24`, the colors come out wrong. 

(My specific use case: drawing the `widpedom` progress by filling a circle and then erasing part of that circle. Currently this will briefly flash the whole circle, so I figured drawing into an ArrayBuffer first would prevent that. Using `Bangle.setLCDMode("doublebuffered")` didn't seem like an option for a widget.)

nebbishhacker, 2020-04-11:
Looks like there's a bug in `asImage()` - the output contains the first byte of each pixel twice:

```
>let b = Graphics.createArrayBuffer(1,1,16).setPixel(0, 0, 0xF800);
=Graphics: {
  buffer: new Uint8Array([0, 248]).buffer
 }
>b.asImage();
={ width: 1, height: 1, bpp: 16,
  buffer: "\xF8\xF8"
 }
```

You can get around that by passing the buffer directly, but then you run into a byte ordering issue - `b` stores the color of each pixel least significant byte first, but `g` does its rendering most significant byte first.

```
let b = Graphics.createArrayBuffer(30,20,16);
b.setColor(0xF800).fillCircle(5, 5, 4); // Should be red, ends up blue with a tiny bit of green
b.setColor(0x07E0).fillCircle(15, 5, 4); // Should be green, ends up red with a tiny bit of blue
b.setColor(0x001F).fillCircle(25, 5,  4); // Should be blue, ends up green with a tiny bit of red
b.setColor(0x00F8).fillCircle(5, 15, 4); // Ends up red
b.setColor(0xE007).fillCircle(15, 15, 4); // Ends up green
b.setColor(0x1F00).fillCircle(25, 15, 4); // Ends up blue
g.drawImage({width: 30, height: 20, bpp: 16, buffer: b.buffer}, 10, 10);
```

The most fool-proof workaround is probably to use a `bpp` of 1, 2, 4, or 8 (depending on how many colors you need), thus sidestepping both issues.

gfwilliams, 2020-04-14:
Thanks - looks like there may be an issue with the code that handles bpp>8. This is definitely an Espruino firmware issue - I'll move this issue over to keep track of it.

**However** `asImage` creates a complete copy of the image right now, so isn't that efficient. If you want to efficiently buffer your drawing I'd suggest skipping it completely and doing:

```
let b = Graphics.createArrayBuffer(10,10,bpp);
b.setColor(0xF800) // Red
b.fillCircle(5,5,4)
let i = {width:10,height:10,bpp:bpp,buffer:b.buffer};
g.drawImage(i,10,10) // put the red circle on the display
```