espruino/BangleApps #277: New app idea - Sleep tracker

brainfart-be, 2020-04-10:
Hi !
I use to wear a fitbit and i liked the sleep tracker feature.

I'm going to try to build one myself.
I think i will need some help to build that so don't hesitate to reply bellow if you are interested!

Well time to learn how a sleep tracker work :)

Purple-Tentacle, 2020-04-11:
I am interested as well.
Currently I am using SleepAsAndroid on my phone. This has the ability to connect to a Google Wear Watch to track movement. 

I do not think that this is possible with Bangle.js. 

But whatI would like
- Start and end of sleep
- Rating of sleep quality (manual, 5 stars)
- Movement detectio 
- Local logging of data
- In the long run: Support for Gadgetbridge, to store and show data on the phone

brainfart-be, 2020-04-11:
Since i have no clue of how any of this work, i was planning to wear my fitbit and my bangle together.

The bangle app would simply record heart rate and accelerometer (or any other usefull sensor)

Then build a webpage to extract those data from the bangle and try to get the same result as the fitbit.

I obviously have to do some research on this topic

Purple-Tentacle, 2020-04-13:
Maybe some of the code I used in Active Pedometer could be used in a sleep tracker app?
Many movements are currently counted as steps. We could use use to detect movement.
We have

- Minimum and maximum time between movements
- Threshold of how much movement has to happen before we go into 'movement' mode
- Timer to reset from 'movement' to 'still' mode

Didgeridoohan, 2020-04-13:
Gonna post this here, just for the record:
https://en.wikipedia.org/wiki/Actigraphy

nxdefiant, 2020-04-29:
Hi,

I wonder if a "Sleep Phase Alarm clock" is part of the idea here. I have an Axbo which can wake me up as much as 30 minutes before my wake up time. It does that by estimating by sleep phase from an accelerometer at my wrist.

But I agree that we definitely need some real data first. Any progress on this?

brainfart-be, 2020-04-29:
Hi,

My initial goal was just to display a graph describing my sleep (like my fitbit does, i should post a screenshot of the graph)

I still didn't start anything yet, i kind of took a break from "code"

As a started we could simply "clone" the heart rate monitor app and add other stuff to track like accelerometer and temperature

Purple-Tentacle, 2020-04-29:
I am currently coding a simple app based on my Active Pedometer app. 
It will simply detect when a certain number of "steps" have been registered in a certain time. 

Very basic, but it can give an estimation of sleep quality based on movement.

nxdefiant, 2020-04-29:
hmm. I was more thinking about a widget that sends raw accel (and maybe pulse) to a remote computer. The data can later be analyzed.

brainfart-be, 2020-04-29:
I think the best way to handle this is to save everything on the watch then download the file on you browser to run computation on it.

I guess being connected on the bluetooth is power consuming and your watch may run out of battery during the night or if the computer shutdown for some reason or you lose connection with it that mean you lose the data.

The logic will be the same, logging data then analyse them later on a computer

nxdefiant, 2020-04-30:
I'm not sure that I agree here. Constantly writing to flash will wear it out fast.

nxdefiant, 2020-04-30:
I made a tiny widget which does sends the data over bluetooth. Problem right now is that I'm too stupid to receive the data on a Linux host, so maybe writing to flash is not so bad afterall...

My problem is the amount of data, my goal is right now to send data at 10Hz. This does obviously produce a lot of data. The Bangle.js has 64kB RAM and 4MB Flash. Assuming we want to use 1MB per night (8h) I get:
> 1MB*10Hz/8h = (1*1024*1024)/(10*60*60*8) = 3 bytes

I can't save the measurement with 3 bytes, so any ideas here?

Note: I currently only want to save the measurements for analysis of what can be done, not for long term usage.

gfwilliams, 2020-05-01:
What are you measuring at 10Hz? Obviously some stuff like heart rate changes at *most* once a second.

So.. You could store it it packets. For each second you could store:

* 1 byte heart rate
* 1 byte temperature
* 10 bytes acceleration magnitude

Which should easily get you down to the level you need?

Long term I'd suggest trying to do a lot more work on the Bangle to reduce the amount of data sent though!

Also: Acceleration comes in by default at 12.5Hz so to avoid aliasing I'd just store all the data you get. It'd be easy enough to just do everything on the acceleration handler (when you get 10 readings save, and so on).

nxdefiant, 2020-05-01:
I need the full acceleration data (x,y,z) only for primary analysis of the data. Later we can decide if e.g. magnitude is enough. ok, 12.5Hz it is, I already wondered about the rate.

Btw later storing full data at 13Hz is probably not needed, just the changes between times with little/much movement

nxdefiant, 2020-05-06:
Status update:
I have now used bluepy and a slightly modified version of [bluepy_uart.py](https://www.espruino.com/Interfacing#python) to receive the acceleration data of a single night. Unfortunately I had lots of bluetooth dropouts, so data is lost, but maybe it is enough for some plotting.

nxdefiant, 2020-05-12:
Did anyone learn something by now, e.g. from [papers](https://arxiv.org/pdf/1802.09358)? I'm currently looking at my recorded data but was not able to make much sense of it yet.

nxdefiant, 2020-05-28:
For info: I have pushed  the alarm clock in #459. The paper ["Towards a Benchmark for Wearable Sleep Analysis with Inertial Wrist-worn Sensing Units"](https://ubicomp.eti.uni-siegen.de/home/datasets/ichi14/index.html.en) is a good introduction into this topic. I have implemented their ESS algorithm in JS for my alarm clock.

If you guys are still interested in painting a graph of sleep states you could use this application as a starting point. Be aware that given the sensors of the Bangle.js you can only get en estimate, not the exact sleep states.

If no one is still interested in this topic I recommend this issue to be closed.

kuleszdl, 2020-07-30:
@nxdefiant It would be awesome if this could be used to record the data to make it fetchable from GadgetBridge as well. However, I assume your alarm clock does not store this data, yet?

nxdefiant, 2020-07-30:
Correct, but storing the data should not be a problem. If you want to store the data there are basically two states in "swest", Awake (true) and Sleep (false). When you store the change from awake to sleep you have to predate the timestamp by 10 minutes (the setting of sleepthresh).

dapgo, 2020-09-06:
is the any plan to develop a sleeping tracker/monitor reusing code  from [SleepPhaseAlarm](https://github.com/espruino/BangleApps/tree/master/apps/sleepphasealarm)?

A sleep tracker is a very useful and common function in smartwatches

DDDanny, 2021-12-14:
Hi, I also wondered about the emtpy "sleep diagramms" and understood it is currently not implemented.
Let me short share an approch with you.

I understood so far:
1. from hardware event the movement diff (not the raw data) of last 10 minutes is already delivered
https://github.com/espruino/Espruino/blob/0a9f07a0fc21d1f989adaa8f35f74b29349307d6/libs/banglejs/jswrap_bangle.c#L2648

2. Bangle js app android has currently only one activity "act" and does neither consider nor handover this movement diff
https://github.com/espruino/BangleApps/blob/253abf53c05a160ce1fa70fd1b33d1268a4ac022/apps/android/boot.js#L58

3. According to this gadgetbridge bangle BangleJSSampleProvider only knows one activity
https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/e6c3ec24ecfae4a51464db05a151738e78fb883c/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/devices/banglejs/BangleJSSampleProvider.java#L38
which is currently "act" (mapped in BangleJSDeviceSupport https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java)

4. As far as I see from other gadgetbridge (device/service/activity) implementations sleep activity/ies has to be implemented

So the appraoch would be to consider in bangle js the movement diff to decide sleep/awake in 2. by using the [sleepphasealarm calc_ess](https://github.com/espruino/BangleApps/blob/253abf53c05a160ce1fa70fd1b33d1268a4ac022/apps/sleepphasealarm/app.js#L15). New bangle js activity type "sleep" is handoverred to gadgetbridge with the heart-rate as it is (so the best read) -> although i would prefer the best min and max read from the last 10 minutes... (i have seen that the raw data is now accessible again)

In gadgetbridge: Based on the "history" on the decission can be made to e.g. devide into 2 or 3 sleeptypes:
* optional: NAP_SLEEP: a period of 10 minutes without movement
* LIGHT_SLEEP: 3 NAP_SLEEP periods
* DEEP_SLEEP: LIGHT_SLEEP and gadgetbridge personal settings like my typicall bpm matches e.g. 50-70

Please give me a feedback. I'm a bit unsure the sleep type decision should also be made in the bangle js (then it could also be reused by others)

Outlook: Once done based on this sleep infos also the the "supportsSmartWakeup" could be be implemented in gadgetbridge.

Have a good day

gfwilliams, 2021-12-14:
Hi - yes, this sounds like a great plan. I'd say keeping track of sleep type in the Bangle would be good - it's easy to update the algorithm and others could use it too.

I was a bit unsure how to work out the sleep periods, but movement was put in there precisely so we *could* do sleep tracking. Are your ideas for detecting NAP/LIGHT/DEEP_SLEEP based on what other devices do, or is it just what you think should work?

khampf, 2021-12-15:
There are provisions in Sleep as Android for monitoring activity data from Pebble using Gadgetbridge, perhaps this could be done on Bangle.js as well. Another approach in Sleep as Android would be to use its DIY option and make an Android app independent of Gadgetbridge. But I have always thought all Mi Bands and similar track movement and the app on the phone analyzes live movement data or by reading accumulated acceleration counters.

DDDanny, 2021-12-16:
1. sleep tracking: I checked the behaviour of other devices and most of them seems to handover the correct Activity State (NOT_WORN; NOT_MEASURED; LIGHT_SLEEP; DEEP_SLEEP) - [ActivityKind.java](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/model/ActivityKind.java) i checked the references and looked at the sample - mappings. I was not able to find the sources.
After thinking more about it it absolutly makes sense to do as much logic in the watch to also allow to handover such data to ios, esp32, rpi or whatever linked BLE device.

2. Sleep As Android / Pebble - it is currently not supported for heart rate, so only movement is tracked.
But yes, there is also an (API at Android via Intents) that can receive movement and hpm.

I'll keep waiting for a working bluetooth dongle beacause using espruino webide on the smartphone is really no fun...

gfwilliams, 2022-02-03:
We do now have a sleep phase alarm app: https://banglejs.com/apps/#sleepphasealarm

Also Bangle.js records a 'movement' value and the 'health' app records it every 10 minutes, so it should be enough for some more detailed sleep calculations if someone wants

closed 2022-02-03 by gfwilliams