espruino/BangleApps #279: Time zone ignore on power up until you re-open morph clock

PaulMcGuinness, 2020-04-11:
If you start the device the clock shows 1am, then syncs to the phone / gps and shows 6:15am (as an example), and won't show 7:15am till you've pressed BTN2 and selected Morph Clock again.

gfwilliams, 2020-04-14:
Where are you syncing time from? Gadgetbridge, or the App loader?

PaulMcGuinness, 2020-04-14:
No idea - How can I tell

PaulMcGuinness, 2020-04-16:
@gfwilliams - Ok, more testing, and I don't think it has anything to do with time syncing.

Here is the scenario:-

My locale is set to +1 (BST)
LCD Wakeup set "On Twist" to "On"

Put watch on desk. Leave for >1 hr pick up and screen comes to life showing Morphing Clock but shows UTC time (+0 hrs) 

Press BTN2, select Morphing Clock again and correct BST (+1) shows.

gfwilliams, 2020-04-16:
Does the clock 'morph' to the new time at all after a second or two?

It's set up so that when you power the screen on, it shows the old time, then morphs to the new one...

PaulMcGuinness, 2020-04-16:
It morphs to the "wrong" time, then when you press BTN2 and go back to it, it just shows the "right" time.

gfwilliams, 2020-04-16:
Are you sure all the software is up to date? This used to be an issue with very early bootloader versions but it was solved at least a month ago.

PaulMcGuinness, 2020-04-16:
Definitely. Just in case there are some gremlins in the system, how can I do a complete 'wipe and factory reset' to a total 'clean' Bangle?

gfwilliams, 2020-04-16:
If you go to https://banglejs.com/apps/, then `About`, and `Install Default Apps`, that should erase everything and start again

PaulMcGuinness, 2020-04-16:
I'll try that later and let you know if it solves it.

PaulMcGuinness, 2020-04-17:
Seems to have fixed it. I suspect it was an issue with legacy bits of code / settings during updating rather than uninstalling / re-installing.

PaulMcGuinness, 2020-04-20:
Ok - This has nothing to do with TimeZones. I've just replicated this from a 'Wiped' device and attached a video to demonstrate the issue.

Essentially;

Let display go to sleep.
Pickup device and move around a bit so screen lights up.
Clock shows time it went to sleep.
Press BTN2 x 2  and correct time shows.

 ([bangle_test.zip](https://github.com/espruino/BangleApps/files/4504759/bangle_test.zip)

PaulMcGuinness, 2020-04-20:
Just tested it again, and it seems the morph freezes when the screen wakes up (by motion)

[image]

gfwilliams, 2020-04-21:
Wow, interesting. So literally just a fresh install by 'Install default apps' (nothing else) and wake up with motion and it hangs mid-morph? Does it happen all the time, or just sometimes? I tried once or twice here with a fresh install and couldn't reproduce it.

PaulMcGuinness, 2020-04-21:
The mid-morph has only happened once. But the "Show the last time" happens every time - Even if I wake it up by pressing BTN1. I did the "Install default apps", but that just seems to cycle an uninstall / install process. Is there a more 'proper' way to factory reset the firmware (without taking it apart and using pogo-pins etc.)

gfwilliams, 2020-04-21:
Install default apps should do it properly - it totally clears the whole flash area then re-installs everything. Updating the watch firmware may help though: https://www.espruino.com/Bangle.js#firmware-updates

PaulMcGuinness, 2020-04-21:
Just had the same frozen morph after blowing new FW onto it :-(

Is it something to do with the 'Twist' screen wakeup?

Didgeridoohan, 2020-04-21:
I just realised I've been seeing something similar with the binary clock the past couple of days. Once in a while when the screen wakes up, the clock is frozen and need a refresh (long press BTN3) to start ticking again.

Will take a closer look now...

commit bf5bf919 (2020-04-22): mclock: Improve performance, attempt to remove occasional glitch when LCD on (fix #279)
closed 2020-04-22 by gfwilliams