espruino/BangleApps #285: [Feature request] Gadgetbridge and bluetooth widgets

Didgeridoohan, 2020-04-12:
I'm still learning JavaScript (and time is not something I have in abundance) or I would have looked into possibly fixing the following myself. But, here goes...

**First of all**, the Gadgetbridge and bluetooth widgets do not update their status on connect or disconnect. Manually redrawing the watch face (either by long-pressing BTN3 or entering and exiting the launcher) is necessary for the widget status to be updated. Is it possible to have them update automatically?

I did notice that @paulcockrell's MarioClock updates the phone connection successfully on both connect and disconnect without manually redrawing the watch face (with a nice little buzz telling you that the connection status changed). I looked a little closer and saw that there seems to be some differences in how the two apps handle the connection changes (but as already stated, analysing the differences is not yet part of my skill set).

**Second**, the Gadgetbridge widget currently seems to only check for a bluetooth connection and not specifically for a connection with the Gadgetbridge app. Would it be possible for it to show as active only when the GB app has a connection with the watch?

**Third and last**, could the Bluetooth widget reflect if BLE is on or off? Would be nice if the bluetooth icon didn't show at all when BLE is turned off.

gfwilliams, 2020-04-14:
> Gadgetbridge and bluetooth widgets do not update their status

Are you sure about this? There's definitely code in there for it: https://github.com/espruino/BangleApps/blob/master/apps/widbt/widget.js#L16

Often phones/PCs hang onto the connection for a few seconds after Web Bluetooth disconnects, which may cause the issue?

> Would it be possible for it to show as active only when the GB app has a connection with the watch?

I'm afraid it's not aware of any difference. Perhaps when it receives the first command from Gadgetbridge it could update though?

> Would be nice if the bluetooth icon didn't show at all when BLE is turned off.

That sounds like a really neat idea. Like charging - no point wasting space if it's not connected.

Didgeridoohan, 2020-04-14:
> > Gadgetbridge and bluetooth widgets do not update their status
> 
> Are you sure about this? There's definitely code in there for it: https://github.com/espruino/BangleApps/blob/master/apps/widbt/widget.js#L16

Yes. I did see that code and wondered about why it does not work.

I have noted this since I got the watch a couple of weeks ago... I just tested it and the watch has stayed connected to my phone for several minutes (5-10) and even received notifications during that time without the Bluetooth or Gadgetbridge widgets showing the connected status.

And, like I said, the MarioClock updates the status just fine...

gfwilliams, 2020-04-15:
Ahh, thanks - just checked and fixed the connection status for both widgets :)

Didgeridoohan, 2020-04-15:
Thanks! It's working as expected now.

closed 2021-02-02 by gfwilliams