espruino/BangleApps #286: How do I include a font module in a published app

fredericrous, 2020-04-13:
I created an app that uses the font module Font7x11Numeric7Seg.
It works great when I run the app from the IDE with send to RAM.

However if I include the app in the App Loader, download it and execute it, I get a black screen.
The IDE logs an issue. It cannot find the module Font7x11Numeric7Seg

```
Uncaught Error: Module Font7x11Numeric7Seg not found
 at line 9 col 30
require('Font7x11Numeric7Seg').add(Graphics);
                             ^
```

What is the procedure to include a font module in an app published on the app loader?

fredericrous, 2020-04-13:
Is the following method the right way => https://github.com/espruino/BangleApps/pull/287/files#diff-a11413622983ee5edc05e448165e396aR9 ?

gfwilliams, 2020-04-14:
Right now (and especially for a small font like this) I think your method is spot on.

Another option is to include it as a separate file and actually write it to a Storage File called `Font7x11Numeric7Seg`, but I think that's overkill here.

Eventually I think we might add a better way of doing this though

gfwilliams, 2021-02-02:
Actually now, modules should get included automatically - you just have to reference them (the code from the first post will work now)

closed 2021-02-02 by gfwilliams